erdos-project/pylot #8: Imprecise vehicle and pedestrian bounding boxes in perfect detector

ICGog, 2019-06-27:
Carla provides 3d world bounding boxes for pedestrians and vehicles. We transform these bounding boxes into 2d bounding boxes which are into the field of view of the RGB camera. However, when we compute the 2d bounding box we take the furthest apart points [https://github.com/erdos-project/pylot/blob/master/pylot/simulation/utils.py#L576](https://github.com/erdos-project/pylot/blob/master/pylot/simulation/utils.py#L576) . Instead, we should use the orienatation of the pedestrian to select the correct plane, and thus provide more accurate 2d bounding boxes.

See for more details on Carla bounding boxes: [https://carla.readthedocs.io/en/stable/measurements/](https://carla.readthedocs.io/en/stable/measurements/)

See bounding box we create:
[image]

sukritkalra, 2019-11-10:
The recent commit (56e5286) fuses segmentation and depth arrays to provide better bounding boxes than before. The following issues still remain:
- [ ] Motorcycles and Riders are not visible in the current commit. We need to flow the vehicle blueprint as part of the GroundVehiclesMessage and provide a different segmentation threshold to the `get_2d_bbox_from_3d_box` for proper bounding boxes.
- [ ] The low depth thresholds (default 5 metres) makes it difficult for us to provide bounding boxes for cars that are visible but are mostly hidden behind other cars. The following picture shows this issue:
[image]

sukritkalra, 2019-12-18:
Fixed both of the above issues in 340957c. However, we now detect extra bounding boxes if the vehicle is in front of us, but occluded by another vehicle. We should do a forward_vector check to remove the heuristic introduced in 340957c, if the vehicle is on the same lane.

sukritkalra, 2020-01-21:
They should be better now. Closing until we need to make further improvements.

closed 2020-01-21 by sukritkalra