erdos-project/pylot #28: Speed limit sign detection code is too slow.

sukritkalra, 2019-11-10:
The `get_speed_limit_det_objs` in `pylot/simulation/utils.py` takes too long (upwards of 2 seconds) to detect speed limit signs in an image. This considerably slows down our data collection capabilities and needs to be fixed.

We should investigate if its possible to put bounding boxes on the speed limit signs by looking at their static meshes in Carla (similar to Traffic light bounding boxes) and deprecate this function.

ICGog, 2020-01-21:
@alvkao58 is looking into this issue. The method is too slow because for each invocation we transform a depth frame into a point cloud (https://github.com/erdos-project/pylot/blob/master/pylot/perception/depth_frame.py#L45). Moreover, as part of the transformation we convert back and forth between numpy arrays and lists of `Locations`.

ICGog, 2020-03-03:
This issue is affecting Pylot performance in the general case as well. It is extremely slow to transform the rich CARLA point clouds into `pylot.util.Locations` and back (see the image below; the bottleneck is the Lidar operator https://github.com/erdos-project/pylot/blob/master/pylot/drivers/carla_lidar_driver_operator.py#L64). IMHO, we should move back point clouds to be numpy array, which we only manipulate through methods defined on the PointCloud class.

[image]

closed 2020-03-11 by ICGog