erdos-project/pylot #66: Improve lane detection

ICGog, 2020-02-23:
We need lane detection for when we don't have access to a fully detailed HD Map. To address this limitation, we can either extract lanes from segmented frames, or integrate Lanenet into Pylot (or any better alternative). We've previously tested Lanenet with [this script](https://github.com/erdos-project/pylot/blob/e4b44b5a2e552dd48aec049f4d1fd28973828fe9/scripts/test_lanenet_lane_detection.py), but we haven't wrapped it in an operator.

sukritkalra, 2020-02-23:
I can take a look at how good LaneNet works with real data from Vijay’s camera frames. Is there any other reason why we would need complete semantic segmentation?

ICGog, 2020-02-23:
I don't really see any other reason at the moment.

ICGog, 2020-07-03:
Lanenet works pretty well, but its postprocessing code is slow (around 420 ms). In particular, https://github.com/ICGog/lanenet-lane-detection/blob/master/lanenet_model/lanenet_postprocess.py#L163 takes around 350 ms. 

We'll have to investigate this if we end up using LaneNet heavily.