erdos-project/pylot #131: Issue with running docker container

wuxiaohua1011, 2020-07-19:
Hi 

I am trying to follow the Deploy using Docker tutorial. I encountered some hiccup and evetually stucked on an error. 

The small hiccups are as follows, maybe it can help improve documentation:
- The command listed on the instruction `ssh -p 20022 -X erdos@localhost /bin/bash` will cause my command line to just hang there, therefore I just did `ssh -p 20022 -X erdos@localhost` and then when I am in the server, I manually execute `/bin/bash`, does this change affect anything?

- Also, it seems like PyGame is not installed by default in this docker image, I manually installed it using `pip3 install pygame`. 

----------------------------------------------------------------------------------------------------------------------------------------------------------------------
The error is described below: 
The error log is attached [here](https://gist.github.com/wuxiaohua1011/605d9681a81a7f1933808b5cc2225bbc)

Have you guys happened to encounter this? It seems like my GPU has ran its ram? From the `nvidia-smi` command, it seems like python is taking most from it? But I am not running any other program besides this one. 

My system specification are as follows:
OS: Ubuntu 18.04
Graphics Driver: NVIDIA-SMI 450.36.06    Driver Version: 450.36.06    CUDA Version: 11.0 
GPU: Nvidia RTX 2070 Super, 8GB memory

Here's the output from `nvidia-smi`
[image]

Here's screenshot of the the window for confirming Xauth is setted up correctly, carla server is running, and the terminal for running `python3 pylot.py --flagfile=configs/detection.conf --visualize_detected_obstacles`. 
[image]

Please let me know if any other info would be necessary. 

Thank you!

wuxiaohua1011, 2020-07-19:
yeah, and i am pretty sure that it is the it is the `frenet_optimal_trajectory_planner/FrenetOptimalTrajectory` that is acting weird. 

Here's my GPU usage before running `python3 pylot.py --flagfile=configs/detection.conf --visualize_detected_obstacles`
[image]

When I execute  `python3 pylot.py --flagfile=configs/detection.conf --visualize_detected_obstacles`, the program froze for a second with this error message
[image]

Then it went to on to complain about `trying to create rpc server for traffic manager; but the system failed to create because of bind error.`
[image]

And somehow my GPU usage is ramped up to full
[image]

pschafhalter, 2020-07-19:
To fix the "Attempted to construct callback over more than 15 read streams bug", try the following:
1. Go to the ERDOS folder.
2. Find and replace 15 with 16 in `build.rs`.
3. Run `python python/setup.py install --user` to rebuild ERDOS.

I think you can fix the other errors with the following:
1. SSH into the docker container.
2. Run `/bin/bash`
3. `cd` into the Pylot directory.
4. Run `export PYLOT_HOME=$(pwd)`

Thanks for reporting These are issues we're aware of and are planning to fix before our upcoming release which should be more stable (sometime next week). CC @ICGog because I ran into these too when testing out the docker container last week.

wuxiaohua1011, 2020-07-20:
hmm, Did you do anything else? 

i tried to replace 15 with 16 in `build.rs`. There are two places where the number 15 appears, one under the `make_add_watermark_callback` function, the other under the `make_callback_builder` function. 

I tried changing 15 -> 16 for both of them, and then for one of them respectively. They all resulted in different kinds of build errors. 

This is the error for changing 15 -> 16 for both of them
[image]

This is the error for changing 15 -> 16 for watermark function. 
[image]

And here's the error for changing 15 -> 16 for  `make_callback_builder`
[image]

pschafhalter, 2020-07-20:
This should be fixed once erdos-project/erdos#132 is merged. In the meantime, you can add `16: "Sixteen"` to the `num_to_str` dictionary in `scripts/make_callback_builder.py`.

ICGog, 2020-07-20:
Thanks for opening the issue. We've added a dependency on pygame, and merged in Peter's fix. Can you please try to:
pull erdos in the container: `cd ~/workspace/erdos ; git pull ; python3 python/setup.py install --user` ?

wuxiaohua1011, 2020-07-21:
The build of ERDOS is successful

however, the command `python3 pylot.py --flagfile=configs/detection.conf --visualize_detected_obstacles` still errored out at loading the `build/libFrenetOptimalTrajectory.so`, here's the [gist](https://gist.github.com/wuxiaohua1011/49f23e4a719f6a974eb8d43c1c489578) for the error log.  
It seems to be complaining about 
`OSError: build/libFrenetOptimalTrajectory.so: cannot open shared object file: No such file or directory`

This file does exist in `~/workspace/pylot/dependencies/frenet_optimal_trajectory_planner/build` but it seems like the name has been referenced wrong in `frenet_optimal_trajectory_planner/FrenetOptimalTrajectory/fot_wrapper.py`

Some special notes: 
1. I had to install pygame manually,  I did that using the command `pip3 install pygame`
2. for adding public keys to access remote servers, the listed instruction in the readme is slightly off, shouldn't it be `pylot:DIRECTORY` instead of `pylot_new:DIRECTORY`
```
# readme
nvidia-docker cp ~/.ssh/id_rsa.pub pylot_new:/home/erdos/.ssh/authorized_keys
nvidia-docker exec -i -t pylot_new sudo chown erdos /home/erdos/.ssh/authorized_keys

# should be
sudo nvidia-docker cp ~/.ssh/id_rsa.pub pylot:/home/erdos/.ssh/authorized_keys
sudo nvidia-docker exec -i -t pylot sudo chown erdos /home/erdos/.ssh/authorized_keys
```

ICGog, 2020-07-21:
Can you check if you have the PYLOT_HOME environment set to the pylot directory in your container? That should be enough for Pylot to find the Frenet file.

Another option is to `git pull` Pylot, which now shouldn't import the Frenet planner by default.

pschafhalter, 2020-07-21:
I had to restart everything when this happened. This is the command I used:

`ps -ef | grep 'pylot' | grep -v grep | awk '{print $2}' | xargs -r kill -9`

ICGog, 2020-07-22:
We've pushed out the ERDOS and Pylot 0.3. Can you please try again our docker image? Let us know if you run into any issues.

wuxiaohua1011, 2020-07-23:
Hi, yes, it works perfectly. I just deleted the image, and then re-pulled the entire thing following the instruction provided in the readme, it worked without any hiccups. 

However, as #133 , it is also pretty laggy (but actually better than the manual installation)

closed 2020-07-23 by wuxiaohua1011