erdos-project/pylot #132: erdos has no attribute called 'profile_method'

wuxiaohua1011, 2020-07-19:
Hi, after failing to run pylot with Docker, I am attempting to run it with source install. Everything went well until the step of actually running it. 

It has given me [this error output ](https://gist.github.com/wuxiaohua1011/e9e13192ca55f49c608fea283fd8d098)

I am wondering how should i go about fixing it?

Best,
Michael Wu

ICGog, 2020-07-20:
This seems to indicate that ERDOS didn't install successfully. Can you please run one of the example ERDOS dataflow programs to see if they work? For example,  https://github.com/erdos-project/erdos/blob/master/python/examples/simple_pipeline.py

wuxiaohua1011, 2020-07-21:
it did not work, it seems like ERDOS did not install correctly? I installed erdos using PIP

[image]

ICGog, 2020-07-21:
Ah! That's the reason! The ERDOS PIP 0.2 version doesn't work with the master Pylot version, only with Pylot 0.2. We're going to do 0.3 releases of ERDOS and Pylot this week, but if you want to deploy before we release the new version you can either:
1) Install ERDOS manually (https://github.com/erdos-project/erdos#rust-installation), or
2) Update ERDOS in the container: ```cd ~/workspace/erdos ; git pull ; python3 python/setup.py install --user ```

wuxiaohua1011, 2020-07-23:
Hi, thanks for the lightning speed of updates! 
I think you've already released ERDOS 0.3, and therefore I just did a 
`pip uninstall erdos` and then `pip install erdos` to update it. 

There is a window that popped up with tracked objects, however it is quite slow and laggy. What I observe is that it run fine for a few frames, and then freeze, and then ran a few more frames. But this will be a seperate issue, therefore, let me just talk about it in other thread

Thanks again for the lightning speed of updates, i can confirm that 0.3 release of ERDOS does fix this thread :)

closed 2020-07-23 by wuxiaohua1011