erdos-project/pylot #133: Simulation is very laggy

wuxiaohua1011, 2020-07-23:
Hi,
I am finally able to run PyLot. 

I found that the simulation is very laggy, especially when starting. And it freezes after a couple smooth frames

I've recorded a short clip to demonstrate the issue: [Youtube link](https://youtu.be/O_NdJ0G6ouQ)

I am pretty sure that I am using my GPU
[image]

My system specification is as follows:
OS: Ubuntu 18.04
GPU: RTX 2070 Super 8GB memory
CPU: Ryzen 3700X

As you can see in the video, im not running any other processes besides this. It would be great if you could provide a video demo. 

Also I observe that there is this constant stream of error, that did not lead the program to crash, but might be alarming
[image]

ICGog, 2020-07-23:
Please let us know what configuration file you're using to run pylot.

wuxiaohua1011, 2020-07-23:
I am running `configs/detection.conf`

ICGog, 2021-03-07:
The simulation is lagging because the detection.config file create many agents in the simulation, and also because the default detection model is too big for you GPU. Please try to use a smaller model, and as well reduce the number of agents.

closed 2021-03-07 by ICGog

littlerants, 2023-05-12:
How big is the memory of your computer？

pschafhalter, 2023-05-17:
Our setup has:
- 2× Xeon Gold 6226 CPUs
- 128GB of RAM 
- 2× Titan-RTX GPUs (24 GB of GPU memory each).

The exact resource requirements depend on the configuration of Pylot.