erdos-project/pylot #134: Processes do not "kill itself" after termination

wuxiaohua1011, 2020-07-23:
Hi, I am still a noob to how processes work but I know that in the original Carla, when i want to quite the client, I can just do a `ctrl+c` in the terminal that i am running the client. 

I did the same thing here, and although the ERDOS/Pylot process did end, the Pygame window does not close.

I've also looked in to my GPU using `nvidia-smi`, it seems like python is also still running. Is there anyways to "chain up" the termination command? 

[image]

Best,
Michael

ICGog, 2020-07-23:
You can kill them with `pkill --signal 9 -f  pylot.py`. Alternatively, you can update ERDOS & Pylot to the latest commit to get Pylot to gracefully shutdown on sigint.

closed 2020-07-23 by ICGog

JCSadeghi, 2023-01-04:
I am finding on the latest commit on master this issue is still occurring. I would be grateful if someone could confirm if they are also having this issue or if I have a problem with my setup (running in the docker container). For my application it would be nicer if I didn't have to use pkill as suggested by @ICGog

grefffin, 2025-03-01:
May I ask if you have solved this problem? I'm also confused about this issue，thanks

JCSadeghi, 2025-03-01:
If I remember correctly, I didn't find any solution and had to forcefully kill the processes

grefffin, 2025-03-01:
sad，thank you~