erdos-project/pylot #147: Leaderboard Evaluation Instructions

JCSadeghi, 2020-11-16:
Hi there,

I'm interested in benchmarking pylot in the carla leaderboard. 

Would it be possible to provide some instructions to build `Dockerfile-challenge.master` and run the evaluation?

Kind Regards,

Jonathan

ICGog, 2020-11-16:
Hi Jonathan,

I've added instructions on how to run Pylot on the CARLA leaderboard routes: https://github.com/erdos-project/pylot#carla-autonomous-driving-challenge

Please feel free to re-open this issue if you run into any issues.

closed 2020-11-16 by ICGog

JCSadeghi, 2020-11-17:
Thank you very much, the docker works for me

Kin-Zhang, 2021-12-27:
For self install not docker it should be mentioned at README that the dockerfile.master have to change, like I saw that on pull image:

```bash
########################################################################################################################
########################################################################################################################
############                                BEGINNING OF USER COMMANDS                                      ############
########################################################################################################################
########################################################################################################################

RUN apt-get update && apt-get install -y clang libgeos-dev python-opencv libqt5core5a libeigen3-dev cmake qtbase5-dev python3-tk
RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
ENV PATH="/root/.cargo/bin/:${PATH}"
RUN rustup default nightly
RUN packages='wheel setuptools setuptools-rust fire' && pip install ${packages}
RUN git clone https://github.com/erdos-project/erdos.git && cd erdos && python3 python/setup.py install --user

RUN packages='absl-py cvxpy gdown lapsolver matplotlib==2.2.4 motmetrics numpy<1.17 open3d-python==0.5.0.0 opencv-python>=4.1.0.25 opencv-contrib-python>=4.1.0.25 pillow>=6.2.2 pytest scikit-image<0.15 scipy==1.2.2 shapely==1.6.4 tensorflow-gpu==1.15.4 torch==1.3.1 torchvision==0.2.1 filterpy==1.4.1 numba==0.50.0 scikit-learn==0.20.0 imgaug==0.2.8 nonechucks==0.3.1 cython' \
	&& pip install ${packages}
ENV PYTHONPATH ${TEAM_CODE_ROOT}/:${TEAM_CODE_ROOT}/dependencies/:${PYTHONPATH}
ENV TEAM_AGENT ${TEAM_CODE_ROOT}/pylot/simulation/challenge/ERDOSAgent.py
ENV TEAM_CONFIG ${TEAM_CODE_ROOT}/pylot/simulation/challenge/challenge.conf
ENV CHALLENGE_TRACK_CODENAME MAP
ENV PYLOT_HOME ${TEAM_CODE_ROOT}/

RUN cd ${TEAM_CODE_ROOT}/dependencies/frenet_optimal_trajectory_planner && rm -r build && ./build.sh
RUN cd ${TEAM_CODE_ROOT}/dependencies/hybrid_astar_planner && rm -r build && ./build.sh
RUN cd ${TEAM_CODE_ROOT}/dependencies/rrt_star_planner && rm -r build && ./build.sh

########################################################################################################################
########################################################################################################################
############                                   END OF USER COMMANDS                                         ############
########################################################################################################################
########################################################################################################################

```