erdos-project/pylot #152: Unable to run docker

FitashUlHaq, 2021-01-06:
Hi

I am getting this error while running the docker

pygame 2.0.0 (SDL 2.0.12, python 3.6.9)
Hello from the pygame community. https://www.pygame.org/contribute.html
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  86
  Current serial number in output stream:  87

is there any work around for this?

FitashUlHaq, 2021-01-06:
nvidia-smi output
[image]

ICGog, 2021-01-06:
It looks like this error might be caused by some X issues. Have you followed the steps required to be able to visualize (https://github.com/erdos-project/pylot#visualizing-components)? In other words, run Pylot after first ssh-ing into the container with X forwarding, rather than run after with docker exec.

Can you please give us the command you're running? You could also try to disable all Pylot visualizations to troubleshot if the visualizations are causing this issue (i.e., ensure no --visualize_* flags are enabled).

FitashUlHaq, 2021-01-07:
I have followed these commands (in order)

- docker pull erdosproject/pylot
- nvidia-docker run -itd --name pylot -p 20022:22 erdosproject/pylot /bin/bash
- nvidia-docker exec -i -t pylot /home/erdos/workspace/pylot/scripts/run_simulator.sh
- nvidia-docker cp ~/.ssh/id_rsa.pub pylot:/home/erdos/.ssh/authorized_keys
- nvidia-docker exec -i -t pylot sudo chown erdos /home/erdos/.ssh/authorized_keys
- nvidia-docker exec -i -t pylot sudo service ssh start
- ssh -p 20022 -X erdos@localhost
- cd /home/erdos/workspace/pylot/
- python3 pylot.py --flagfile=configs/detection.conf --visualize_detected_obstacles

FitashUlHaq, 2021-01-07:
If i remove the --visualize_detected_obstacles from the command 
python3 pylot.py --flagfile=configs/detection.conf

I dont see any ouput but the error also disappears

FitashUlHaq, 2021-01-07:
I fixed it by downgrading pygame to 1.9.6 using command pip3 install pygame==1.9.6

closed 2021-01-07 by FitashUlHaq