erdos-project/pylot #154: Lanenet_model not found

FitashUlHaq, 2021-01-18:
Hi, 
I am using the docker to run the pylot and I am getting this error in case of lane detection module

Traceback (most recent call last):
  File "pylot.py", line 272, in main
    node_handle, control_display_stream = driver()
  File "pylot.py", line 150, in driver
    center_camera_stream, pose_stream, open_drive_stream)
  File "/home/erdos/workspace/pylot/pylot/component_creator.py", line 248, in add_lane_detection
    center_camera_stream)
  File "/home/erdos/workspace/pylot/pylot/operator_creator.py", line 221, in add_lanenet_detection
    from pylot.perception.detection.lanenet_detection_operator import \
  File "/home/erdos/workspace/pylot/pylot/perception/detection/lanenet_detection_operator.py", line 20, in <module>
    from lanenet_model import lanenet  # noqa: I100 E402
ModuleNotFoundError: No module named 'lanenet_model'

I have tried changing the pythonpath, adding the path using sys but nothing works.

Any leads regarding this?

closed 2021-01-19 by ICGog

pschafhalter, 2021-01-19:
Hi, thanks for opening the issue! I've opened a PR (#155) to fix this issue -- looks like the import paths were broken while working on lane detection.

While looking into the issue, I also found that some of the paths in our fork of lanenet needed to be updated and opened ICGog/lanenet-lane-detection#1 to address this as well.

Once the PRs are merged, you'd need to pull the latest versions of Pylot and lanenet:
```bash
cd $PYLOT_HOME
git pull
# You may need to re-install Pylot with
python3 setup.py install --user
cd $PYLOT_HOME/dependencies/lanenet
git pull
```

Hope this fixes things! Feel free to reopen the issue if you run into any further problems.

FitashUlHaq, 2021-01-19:
Hi,

Thank you for the quick reply, the error for lanenet is fixed but now I am getting another exception

Traceback (most recent call last):
  File "pylot.py", line 286, in main
    node_handle, control_display_stream = driver()
  File "pylot.py", line 164, in driver
    center_camera_stream, pose_stream, open_drive_stream)
  File "/home/erdos/workspace/pylot/pylot/component_creator.py", line 262, in add_lane_detection
    center_camera_stream)
  File "/home/erdos/workspace/pylot/pylot/operator_creator.py", line 221, in add_lanenet_detection
    from pylot.perception.detection.lanenet_detection_operator import \
  File "/home/erdos/workspace/pylot/pylot/perception/detection/lanenet_detection_operator.py", line 10, in <module>
    from lanenet.lanenet_model import lanenet  # noqa: I100 E402
  File "/home/erdos/workspace/pylot/dependencies/lanenet/lanenet_model/lanenet.py", line 13, in <module>
    from local_utils.config_utils import parse_config_utils
ModuleNotFoundError: No module named 'local_utils'

pschafhalter, 2021-01-19:
Did you pull the update to the lanenet repo? I was getting that error before those changes -- I'm running via the docker container by the way.
```bash
cd $PYLOT_HOME/dependencies/lanenet
git pull
```

If that doesn't work, could you try adding the lanenet folder to the PYTHONPATH?
```bash
export PYTHONPATH="$PYTHONPATH:$PYLOT_HOME/dependencies/lanenet/"
```

FitashUlHaq, 2021-01-20:
yes I pulled the lanenet repo, it got fixed when i added it to pythonpath. I also had to install pyyaml using command

pip3 install pyyaml

now it works.

thank you!

PR #155 (2021-01-19): Fix lanenet imports
Fixes #154 along with ICGog/lanenet-lane-detection#1.