erdos-project/pylot #156: Unable to Run Pylot & CARLA Simulator in WSL 2

simonguozirui, 2021-01-27:
Here is my experience trying to set up Pylot using [Windows Subsystem for Linux (WSL 2)](https://docs.microsoft.com/en-us/windows/wsl/) and issues I met along the way. I hope this can be useful for people trying to do the same thing in the future.

**TL;DR** There seems to be issues with WSL or `nvidia-docker` that Pylot can't be run in WSL 2; would not recommend anyone trying to deploy Pylot on WSL 2 yet.

**Motivation**
I decided to try spinning up Pylot on WSL to utilize my GPU locally instead of running on servers, for easier development and potentially better performance. The recent [NVIDIA's support for CUDA on WSL](https://developer.nvidia.com/blog/announcing-cuda-on-windows-subsystem-for-linux-2/) seems to be a possible way to get this working. 

**Spec and Set Up**
- WSL Version: 2, runs Ubuntu 20.04.1 LTS.
- GPU: NVIDIA GeForce GTX 1060 with 6 GB VRAM
- Windows: Version 2004, OS build 21296.1010 (installed with Windows Insider Program, using Fast Ring/Dev Channel, do this at your own risk)
- X11 forwarding with `VcXsrv` in WSL 2 to show graphics; firewall config changed to allow forwarding.

I mostly followed the instructions using this [CUDA on WSL User Guide](https://docs.nvidia.com/cuda/wsl-user-guide/) on NVIDIA developer zone.  Be very careful what drivers you are installing. This a pretty good guide and by following it I was able to:
- set up special driver and `nvidia-docker` in WSL using latest early preview Windows build
- able to run images (not Pylot) using `nvidia-docker` and get GPU acceleration working. Particular `docker run --gpus all [nvcr.io/nvidia/k8s/cuda-sample:nbody](http://nvcr.io/nvidia/k8s/cuda-sample:nbody) nbody -gpu -benchmark`
- set up X11 and `VcXsrv` to forward any graphics running in WSL to show up
- boot up Pylot docker image in WSL 2

However, the main issue is that the CARLA simulator cannot start successfully. 

When I run `nvidia-docker exec -i -t pylot /home/erdos/workspace/pylot/scripts/run_simulator.sh`, the following output occurs and the program quits **immediately.**

```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```

Turns out these are normal warning flags from CARLA, but the simulator is unable to start successfully.

Since `./run_simulator.sh` is running in headless state, no graphical window would open but it shouldn't be the case that it quits immediately. 

Although the simulator cannot be run, the `pylot.py` scripts still can be run; it will just quit after timing out not finding a working simulator to connect to. 

I tried a few things but none ended up working, including

- Running `./run_simulator.sh` inside the container instead of running through `nvidia-docker exec`
- Running `docker run --gpus` and `docker run -d -it --runtime=nvidia` instead of using `nvidia-docker` as examples shown in the NVIDIA developer guide.
- Not deploy Pylot by Docker, but by manual installation.

I found similar issues on the CARLA repo, notably [issues #2970](https://github.com/carla-simulator/carla/issues/2970) and [issues #3431](https://github.com/carla-simulator/carla/issues/3431); it seems people (both using docker and without using docker) have no success with starting CARLA simulator in WSL yet and faces similar issues as I described. In fact, one of the CARLA's lead developers said they have not tested a WSL setup ([here](https://github.com/carla-simulator/carla/issues/2970#issuecomment-649856931)).

**I believe this is an issue either with CARLA simulator, or `nvidia-docker` and WSL 2's support for CUDA in WSL.** It is hard to debug as `nvidia-smi` does not work in WSL 2 yet ([link](https://forums.developer.nvidia.com/t/hiccups-setting-up-wsl2-cuda/128641/2?u=simonguozirui)). I can't tell whether the CARLA simulator running in docker was able to find the right GPU driver, and based on the Task Manager monitoring, the GPU was never active when I attempt to start the simulator. 

Thanks @pschafhalter for help me debug this! Right now, I don't recommend anyone try run Pylot and CARLA simulator on WSL, as a lot of the WSL-GPU infrastructure has just been released and lack some essential tooling. If you found a way to resolve this, please share it with us!

ICGog, 2021-06-03:
Closing this issue as we won't be able to fix it in the near future.

closed 2021-06-03 by ICGog

NaveedBhuiyan, 2023-02-14:
I am trying to run the carla simulator on wsl2 also in a docker container but no luck. I keep getting the error "no vompatible vulkan found". However when I run the docker container on a linux host everything works just fine. Nvidia-smi works on wsl2 now but looks like it still cant find a compatible vulkan support specially the later carla releases. I am yet not sure which segment is causing the issue, wsl2 or carla itlest

simonguozirui, 2023-08-19:
@NaveedBhuiyan  I tried to run Carla on WSL 2 (with `nvidia-smi` working) in docker too 2 years ago too and had very little luck. I would suggest just running it or the docker on Linux.

simonguozirui, 2025-04-14:
Hi I haven't tried this for a long time since I opened this issue. 
Maybe it would on Windows (more of a CARLA issue), but I would recommend just running this on Linux!