erdos-project/pylot #157: run into OOM for demo.conf

bobd988, 2021-02-18:
Hi,

I am running with native mode. My PC environment: Ubuntu 18.04  64G RAM, Nvidia Driver Version: 460.32.03   CUDA Version: 11.2,  GPU GTX1080. Python 3.6. Carla  was directly started from the  pylot/dependencies/CARLA_0.9.10.1/CarlaUE.sh (not from run_simulator.sh)

I was able to  run the detection  command without error:  python3 pylot.py --flagfile=configs/detection.conf

I0208 21:09:45.181998 140304352266048 __init__.py:409] $HOME=/home/demo
I0208 21:09:45.182323 140304352266048 __init__.py:409] matplotlib data path /home/demo/.local/lib/python3.6/site-packages/matplotlib/mpl-data
I0208 21:09:45.186905 140304352266048 __init__.py:1156] loaded rc file /home/demo/.local/lib/python3.6/site-packages/matplotlib/mpl-data/matplotlibrc
I0208 21:09:45.189215 140304352266048 __init__.py:1879] matplotlib version 2.2.4
I0208 21:09:45.189343 140304352266048 __init__.py:1880] interactive is False
I0208 21:09:45.189993 140304352266048 __init__.py:1881] platform is linux
I0208 21:09:45.190147 140304352266048 __init__.py:1882] loaded modules: ['builtins', 'sys', '_frozen_importlib', '_imp', '_warnings', '_thread', '_weakref', '_frozen_importlib_external', '_io', 'marshal', 'posix', 'zipimport', 'encodings', 'codecs', '_codecs', 'encodings.aliases', 'encodings.utf_8', '_signal', '__main__', 'encodings.latin_1', 'io', 'abc', '_weakrefset', 'site', 'os', 'errno', 'stat', '_stat', 'posixpath', 'genericpath', 'os.path', '_collections_abc', '_sitebuiltins', 'encodings.cp437', 'sysconfig', '_sysconfigdata_m_linux_x86_64-linux-gnu', '_bootlocale', '_locale', 'types', 'functools', '_functools', 'collections', 'operator', '_operator', 'keyword', 'heapq', '_heapq', 'itertools', 'reprlib', '_collections', 'weakref', 'collections.abc', 'importlib', 'importlib._bootstrap', 'importlib._bootstrap_external', 'warnings', 'importlib.util', 'importlib.abc', 'importlib.machinery', 'contextlib', 'mpl_toolkits', 'google', 'zope', 'sitecustomize', 'apport_python_hook', 'signal', 'enum', 'absl', 'absl.app', '__future__', 'pdb', 're', 'sre_compile', '_sre', 'sre_parse', 'sre_constants', 'copyreg', 'cmd', 'string', '_string', 'bdb', 'fnmatch', 'inspect', 'ast', '_ast', 'dis', 'opcode', '_opcode', 'linecache', 'tokenize', 'token', 'code', 'traceback', 'argparse', 'copy', 'textwrap', 'gettext', 'locale', 'struct', '_struct', 'codeop', 'glob', 'pprint', 'absl.command_name', 'absl.flags', 'getopt', 'absl.flags._argument_parser', 'csv', '_csv', 'absl.flags._helpers', 'fcntl', 'termios', 'six', 'six.moves', 'absl.flags._defines', 'absl.flags._exceptions', 'absl.flags._flag', 'absl._collections_abc', 'absl.flags._flagvalues', 'logging', 'time', 'threading', 'atexit', 'xml', 'xml.dom', 'xml.dom.domreg', 'xml.dom.minidom', 'xml.dom.minicompat', 'xml.dom.xmlbuilder', 'xml.dom.NodeFilter', 'typing', 'typing.io', 'typing.re', 'absl.flags._validators', 'absl.logging', 'getpass', 'socket', '_socket', 'selectors', 'math', 'select', 'timeit', 'gc', 'absl.logging.converter', 'faulthandler', 'erdos', 'multiprocessing', 'multiprocessing.context', 'multiprocessing.process', 'multiprocessing.reduction', 'pickle', '_compat_pickle', '_pickle', 'array', '__mp_main__', 'erdos.internal', 'erdos.streams', 'erdos.message', 'erdos.timestamp', 'erdos.operator', 'json', 'json.decoder', 'json.scanner', '_json', 'json.encoder', 'numpy', 'numpy._globals', 'numpy.__config__', 'numpy.version', 'numpy._distributor_init', 'numpy.core', 'numpy.core.info', 'numpy.core.multiarray', 'numpy.core.overrides', 'datetime', '_datetime', 'numpy.core._multiarray_umath', 'numpy.compat', 'numpy.compat._inspect', 'numpy.compat.py3k', 'pathlib', 'ntpath', 'urllib', 'urllib.parse', 'numpy.core.umath', 'numpy.core.numerictypes', 'numbers', 'numpy.core._string_helpers', 'numpy.core._type_aliases', 'numpy.core._dtype', 'numpy.core.numeric', 'numpy.core._internal', 'platform', 'subprocess', '_posixsubprocess', 'ctypes', '_ctypes', 'ctypes._endian', 'numpy.core.fromnumeric', 'numpy.core._methods', 'numpy.core.arrayprint', 'numpy.core.defchararray', 'numpy.core.records', 'num
.....

However when I run demo command I got OOM  error : 

 python3 pylot.py --flagfile=configs/demo.conf 

2021-02-08 21:28:08.108142: W tensorflow/core/framework/op_kernel.cc:1651] OP_REQUIRES failed at conv_ops.cc:886 : Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc

  (0) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[node FirstStageFeatureExtractor/resnet_v1_101/resnet_v1_101/block1/unit_1/bottleneck_v1/shortcut/Conv2D (defined at /usr/local/lib/python3.6/dist-packages/tensorflow_core/python/framework/ops.py:1748) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

	 [[SecondStagePostprocessor/BatchMultiClassNonMaxSuppression/map/strided_slice/_235]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

  (1) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[node FirstStageFeatureExtractor/resnet_v1_101/resnet_v1_101/block1/unit_1/bottleneck_v1/shortcut/Conv2D (defined at /usr/local/lib/python3.6/dist-packages/tensorflow_core/python/framework/ops.py:1748) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

0 successful operations.
0 derived errors ignored.

I have tried another laptop with Ubuntu 16.04 with 64G RAM with CUDA 11 (GTX1070) but got same error. What is minimum hardware required to run this? Thanks.

ICGog, 2021-02-18:
We're usually using more powerful GPUs, but you can modify `detection.conf` to try a few things to get Pylot to run on your machine:

1. You can change `--obstacle_detection_gpu_memory_fraction=0.6` to `0.8` to allocate more of the GPU memory to the detection model. Hopefully, this will be enough for the model to fit.
2. If the above doesn't fix your issue, you can try to use a smaller model by setting `--obstacle_detection_model_paths` flag to `dependencies/models/obstacle_detection/ssd-mobilenet-v1/frozen_inference_graph.pb` or one of the other mobilenet models Pylot downloads upon installation.

Please let us know if this fixes your issue.

bobd988, 2021-02-19:
Thanks.

I tried #1 the log still showed OOM but car is moving and stopped at traffic light when red and continue when green. Does it mean this demo case runs correctly?

Then I tried continued with #2 it becomes worse. the program crashes and no car shows up. Here is the log for #1 and #2

Should I use an more powerful  GPU machine? 

Feb 18 19:18:11.962 WARN 3646 operator events queued in lattice. Increase number of operator executors or decrease incoming message frequency to reduce load.
Feb 18 19:18:11.962 WARN 3646 operator events queued in lattice. Increase number of operator executors or decrease incoming message frequency to reduce load.
2021-02-18 19:18:11.993166: W tensorflow/core/common_runtime/bfc_allocator.cc:419] Allocator (GPU_0_bfc) ran out of memory trying to allocate 64.0KiB (rounded to 65536).  Current allocation summary follows.
2021-02-18 19:18:11.993985: W tensorflow/core/common_runtime/bfc_allocator.cc:424] **************************x************xxxxxxxxxxx**************************************************
2021-02-18 19:18:11.994030: W tensorflow/core/framework/op_kernel.cc:1651] OP_REQUIRES failed at conv_ops.cc:886 : Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1365, in _do_call
    return fn(*args)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1350, in _run_fn
    target_list, run_metadata)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1443, in _call_tf_sessionrun
    run_metadata)
tensorflow.python.framework.errors_impl.ResourceExhaustedError: 2 root error(s) found.
  (0) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[{{node FirstStageFeatureExtractor/resnet_v1_101/resnet_v1_101/block1/unit_1/bottleneck_v1/shortcut/Conv2D}}]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

	 [[SecondStagePostprocessor/BatchMultiClassNonMaxSuppression/map/while/MultiClassNonMaxSuppression/Minimum_6/_275]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

  (1) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[{{node FirstStageFeatureExtractor/resnet_v1_101/resnet_v1_101/block1/unit_1/bottleneck_v1/shortcut/Conv2D}}]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

0 successful operations.
0 derived errors ignored.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/demo/.local/lib/python3.6/site-packages/erdos/streams.py", line 114, in internal_callback
    callback(msg, *write_streams)
  File "/home/demo/.local/lib/python3.6/site-packages/erdos/__init__.py", line 294, in wrapper
    return func(*args, **kwargs)
  File "/home/demo/disk512/github_erdos/pylot/pylot/perception/detection/detection_operator.py", line 115, in on_msg_camera_stream
    msg.frame.frame)
  File "/home/demo/disk512/github_erdos/pylot/pylot/perception/detection/detection_operator.py", line 169, in __run_model
    feed_dict={self._image_tensor: image_np_expanded})
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 956, in run
    run_metadata_ptr)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1180, in _run
    feed_dict_tensor, options, run_metadata)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1359, in _do_run
    run_metadata)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow_core/python/client/session.py", line 1384, in _do_call
    raise type(e)(node_def, op, message)
tensorflow.python.framework.errors_impl.ResourceExhaustedError: 2 root error(s) found.
  (0) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
	 [[node FirstStageFeatureExtractor/resnet_v1_101/resnet_v1_101/block1/unit_1/bottleneck_v1/shortcut/Conv2D (defined at /usr/local/lib/python3.6/dist-packages/tensorflow_core/python/framework/ops.py:1748) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

	 [[SecondStagePostprocessor/BatchMultiClassNonMaxSuppression/map/while/MultiClassNonMaxSuppression/Minimum_6/_275]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

  (1) Resource exhausted: OOM when allocating tensor with shape[256,64,1,1] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc
[... cut ...]
2021-02-18 19:30:22.074143: W tensorflow/core/kernels/gpu_utils.cc:48] Failed to allocate memory for convolution redzone checking; skipping this check. This is benign and only means that we won't check cudnn for out-of-bounds reads and writes. This message will only be printed once.
2021-02-18 19:30:22.076487: W tensorflow/core/common_runtime/bfc_allocator.cc:239] Allocator (GPU_0_bfc) ran out of memory trying to allocate 16.69MiB with freed_by_count=0. The caller indicates that this is not a failure, but may mean that there could be performance gains if more memory were available.
...

ICGog, 2021-03-07:
It looks like your GPU might be a bit too weak for our models. Can you please try with running Pylot with a better GPU, or alternatively download a smaller model from https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/tf1_detection_zoo.md and set `--obstacle_detection_model_paths` to the location of the downloaded `frozen_inference_graph.pb`?

Please feel free to re-open the issue if these suggestions do not fix your issue.

closed 2021-03-07 by ICGog