erdos-project/pylot #161: Metric Implementation

anupamsobti, 2021-03-05:
Hi 

I'm wondering if there are any metrics available for end-to-end evaluation of a drive quality?
[Section 6](https://www2.eecs.berkeley.edu/Pubs/TechRpts/2020/EECS-2020-58.pdf) provides some of these metrics, Collision tables/Distance to vehicles. Are these implementations available?

ICGog, 2021-03-10:
Our [EvalMetricLoggerOperator](https://github.com/erdos-project/pylot/blob/master/pylot/loggers/eval_metric_logger_operator.py) logs infraction information (e.g., intensity of collisions, when lanes are invaded, when traffic lights are not respected,lateral and longitudinal jerk).

In order to ensure that these metrics are going to be logged, please set the `--evaluation` flag to true when running pylot.

Please feel free to re-open this issue if you need additional details.

closed 2021-03-10 by ICGog

anupamsobti, 2021-03-16:
Hi

- I was able to use this to start the additional logging sensor but I'm not sure where to view the summary? I'm able to see some collision sensor reports in the log.
- Also, can I terminate the simulation after an episode of a certain time and view the evaluation at the end?
- For the detectors, I'm able to see the faster_rcnn output, however, I'd like the ground truth boxes to be printed as well. Can you guide me how to achieve this? I'm sharing my flagfile for your reference.
```
######### Detection #########
--obstacle_detection
--obstacle_detection_model_paths=dependencies/models/obstacle_detection/faster-rcnn/frozen_inference_graph.pb
--obstacle_detection_model_names=faster-rcnn
--nosimulator_obstacle_detection
--obstacle_detection_min_score_threshold=0.3
--obstacle_detection_gpu_memory_fraction=0.3
--evaluate_obstacle_detection
--detection_metric=mAP
###### Traffic light detector config ######
--notraffic_light_detection
#--nosimulator_traffic_light_detection
#--traffic_light_det_gpu_memory_fraction=0.3
#--traffic_light_det_min_score_threshold=0.3
###### Depth #####
--perfect_depth_estimation
###### Segmentation config ######
--nosegmentation
#--perfect_segmentation
###### Prediction config ######
--perfect_obstacle_tracking
# --obstacle_tracking
# --tracker_type=sort
--tracking_num_steps=10
--prediction
--prediction_type=linear
--prediction_num_past_steps=10
--prediction_num_future_steps=10
###### Planning config #####
--planning_type=waypoint
###### Control config #####
--control=mpc
######### Carla config #########
--simulator_num_people=50
--simulator_num_vehicles=20
--camera_image_width=800
--camera_image_height=600
# Do not randomly spawn the vehicle.
--simulator_spawn_point_index=1
--simulator_num_people=200
--simulator_num_vehicles=10
--simulator_timeout=60
# Save Data
--data_path=data/
--simulation_recording_file=simrecording/
--log_detector_output
#--log_every_nth_message=10
#--log_rgb_camera
#--log_detector_output
#--log_obstacles
# Evaluation
--evaluation
######### Logging config #########
--log_file_name=pylot.log
--csv_log_file_name=pylot.csv
--v=1
######### Other config #########
--visualize_detected_obstacles
#--visualize_tracked_obstacles
#--visualize_prediction
--visualize_waypoints
#--visualize_rgb_camera
```
Thanks.