erdos-project/pylot #176: Problem running CARLA autonomous driving challenge manually on the machine

FitashUlHaq, 2021-03-26:
Hi,

I am trying to execute the version of Pylot submitted for carla  challenge. It works fine in the docker but I am trying to configure outside docker to be able to make changes in it.

When I execute the last command it fails in step 11. Below is the error I found

Step 11/38 : RUN packages='py_trees==0.8.3 shapely six dictor requests ephem tabulate' 	&& pip3 install ${packages}
 ---> Running in 5ad1799e8506
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 7, in <module>
    from pip._internal.cli.main import main
  File "/usr/local/lib/python3.5/dist-packages/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
Removing intermediate container 5ad1799e8506
The command '/bin/sh -c packages='py_trees==0.8.3 shapely six dictor requests ephem tabulate' 	&& pip3 install ${packages}' returned a non-zero code: 1

any help in this case will be highly appreciated

ICGog, 2021-06-14:
There's no need to run it outside docker because you can also make changes in the docker container, copy files to it, and out of it.

It looks like you're actually trying to build the docker container, rather than install Pylot as this log shows. If that's what you want to do, then probably replacing `packages='py_trees==0.8.3 shapely six dictor requests ephem tabulate' && pip3 install ${packages}` with `pip3 install py_trees==0.8.3 shapely six dictor requests ephem tabulate` should fix your issue.

closed 2021-06-14 by ICGog

SunHaoOne, 2021-06-15:
> There's no need to run it outside docker because you can also make changes in the docker container, copy files to it, and out of it.
>
> It looks like you're actually trying to build the docker [...]

Thanks for your idea. And I have tried it, it shows as follows:
```
Step 11/37 : RUN pip3 install py_trees==0.8.3 shapely six dictor requests ephem tabulate WORKDIR /workspace
 ---> Running in d031bafa7e90
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 7, in <module>
    from pip._internal.cli.main import main
  File "/usr/local/lib/python3.5/dist-packages/pip/_internal/cli/main.py", line 58
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
Removing intermediate container d031bafa7e90
The command '/bin/sh -c pip3 install py_trees==0.8.3 shapely six dictor requests ephem tabulate WORKDIR /workspace' returned a non-zero code: 1

```
Do you have any idea on how to fix it? (I also try to build the docker from the docker file). @FitashUlHaq

FitashUlHaq, 2021-06-15:
> Thanks for your idea. And I have tried it, it shows as follows:
>
> ```
> Step 11/37 : RUN pip3 install py_trees==0.8.3 shapely six dictor requests ephem tabulate WORKDIR /workspace
>  ---> Running in [...]

Hi, 
I was unable to install it on my local machine so I am using docker to run pylot

SunHaoOne, 2021-06-15:
> Hi,
> I was unable to install it on my local machine so I am using docker to run pylot

Thanks for you quick reply. I am now building docker for `carla-leaderboard`, and I will check and try to fix it, thanks again!

SunHaoOne, 2021-06-16:
Finally, I found there is some `pip3` version error with `python3.5`, so after checking, it is neccssary to set `pip3 install --upgrade "pip < 21.0" && \`, and later the terminal shows :

```
DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.
```
And I successfully build the docker image finally.