erdos-project/pylot #182: Vehicle not stopping upon reaching goal destination

FitashUlHaq, 2021-04-08:


FitashUlHaq, 2021-04-30:
Sorry for replying late. I debugged with your suggestion on slack and found out that way-point length does not become 0 upon reaching the destination  (it stays 1) that is why the ego vehicles starts going left and right. I am trying to figure out a way to stop this.

saurabhjha1, 2021-06-13:
Even after updating the planning/waypoints.py as suggested above, the car goes lett or right. Any suggestions ?

saurabhjha1, 2021-06-16:
I fixed this issue through a hack. I am not sure if the hack will work with the challenge scenarios but it works for e2e.conf (tested). Moreover, below code works only when all visualizations are turned off as pygame.quit() does not really kill the pygame as expected. 

The idea is to
(i) calculate the dist = EuclidianDistance(position(EV.current_position) - goal position) by implementing EgoGoalReachTerminate operator, and 
(ii) If the dist < d, initiate graph shutdown by calling shutdown_pylot(). 

I am not doing a pull request as I am not sure if it works for all cases but someone can adapt the below solution for their usage. Also, I am not sure if this is the best way to address the issue as I just started using the erdos framework.

Implementation
---
---
add following operator (possibly in operator_creator.py):
----
```
def add_goal_reach_terminate(pose_stream):
    class EgoGoalReachTerminate(erdos.Operator):
        def __init__(self, pose_stream: erdos.ReadStream,
                     goal_stream: erdos.WriteStream, flags):
            self.pose_stream = pose_stream
            self.goal_stream = goal_stream
            self.flags = flags
            self.closed = False
            pose_stream.add_callback(self.process, [self.goal_stream])
        @staticmethod
        def connect(pose_stream):
            return [erdos.WriteStream()]
        def process(self, pose_msg: erdos.Message, goal_stream: erdos.WriteStream):
            goal_location = pylot.utils.Location(float(FLAGS.goal_location[0]),
                                                float(FLAGS.goal_location[1]), float(FLAGS.goal_location[2]))
            pose_location = pose_msg.data.transform.location
            goal_dist = pose_location.distance(goal_location)
            timestamp = pose_msg.timestamp
            goal_stream.send(erdos.Message(timestamp, goal_dist))
    op_config = erdos.OperatorConfig(name="EgoGoalReachTerminate",
                                     log_file_name=FLAGS.log_file_name,
                                     csv_log_file_name=FLAGS.csv_log_file_name,
                                     profile_file_name=FLAGS.profile_file_name)
    [ego_distance_from_goal_stream,] = erdos.connect(EgoGoalReachTerminate, op_config, [pose_stream], FLAGS)
    return ego_distance_from_goal_stream
```

in pylot.py
---
modify main() to
```
def main(args):
    client, world = get_world(FLAGS.simulator_host, FLAGS.simulator_port,
                              FLAGS.simulator_timeout)
    try:
        if FLAGS.simulation_recording_file is not None:
            client.start_recorder(FLAGS.simulation_recording_file)
        node_handle, control_display_stream, ego_distance_from_goal_extract_stream = driver()
        signal.signal(signal.SIGINT, shutdown)
        while True:
            dist_msg = ego_distance_from_goal_extract_stream.read()
            try:
                if dist_msg.data!= None:
                    print("Distance to goal = {}".format(dist_msg.data))
                    if dist_msg.data < 5.0:
                        shutdown_pylot(node_handle, client, world)
                        exit(0)
            except Exception:
                traceback.print_exc()   
        
    except KeyboardInterrupt:
        shutdown_pylot(node_handle, client, world)
    except Exception:
        shutdown_pylot(node_handle, client, world)
        raise
```

add following to the driver():

`ego_distance_from_goal_stream = pylot.operator_creator.add_goal_reach_terminate(pose_stream)`

and modify return statement of driver to 
`return node_handle, control_display_stream, ego_distance_from_goal_extract_stream`

comment out the following line

`set_asynchronous_mode(world)`