erdos-project/pylot #184: error on runnuing pylot.py

hadeeressamanwer, 2021-04-19:
can anyone help me please
when i try to run pylot.py these errors appear to me 

ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10

RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.

[image]

FitashUlHaq, 2021-04-30:
Advice from fellow user: use docker, it already handles all these errors

ICGog, 2021-06-03:
As Fitash suggested, I advise you to use the Docker image we provide as it already has installed all the required dependencies.

It looks like there's a bunch of issues with your current instalattion:

1. Cuda doesn't find a device. You can check if your GPU is visible by running `nvidia-smi`.
2. The Frenet Optimal Trajectory planner is not installed. Have you executed `./install.sh`?
3. The port specified by carla_traffic_manager_port is already in use. Please try to use a different port.

closed 2021-06-03 by ICGog

Chiakiaki, 2021-09-26:
Hi, I met the same issue. The reason is not setting PYLOT_HOME, as seen in the for_wrapper.py:

try:
    cdll = CDLL("build/libFrenetOptimalTrajectory.so")
except:
    cdll = CDLL("{}/dependencies/frenet_optimal_trajectory_planner/"
                "build/libFrenetOptimalTrajectory.so".format(
                    os.getenv("**PYLOT_HOME**")))