erdos-project/pylot #192: Pylot Failing at round about in town 03

FitashUlHaq, 2021-06-11:
Hi, when PYLOT enters the round about in town03 it starts giving the following exception:

Traceback (most recent call last):
  File "/home/erdos/.local/lib/python3.6/site-packages/erdos-0.3.1-py3.6-linux-x86_64.egg/erdos/streams.py", line 114, in internal_callback
    callback(msg, *write_streams)
  File "/home/erdos/.local/lib/python3.6/site-packages/erdos-0.3.1-py3.6-linux-x86_64.egg/erdos/__init__.py", line 294, in wrapper
    return func(*args, **kwargs)
  File "/home/erdos/workspace/pylot/pylot/perception/detection/lanenet_detection_operator.py", line 117, in on_camera_frame
    lane, msg.frame.camera_setup)
  File "/home/erdos/workspace/pylot/pylot/perception/detection/lanenet_detection_operator.py", line 190, in lane_to_ego_coordinates
    ego_lane = to_world_transform.transform_locations(locations)
  File "/home/erdos/workspace/pylot/pylot/utils.py", line 733, in transform_locations
    transformed_points = self.__transform(points, self.matrix)
  File "/home/erdos/workspace/pylot/pylot/utils.py", line 671, in __transform
    points = np.append(points, np.ones((1, points.shape[1])), axis=0)
IndexError: tuple index out of range

FitashUlHaq, 2021-06-14:
same exception at X roads

ICGog, 2021-06-14:
It looks like you're using the lanenet lane detection operator. Have added code to Pylot to use the detected lanes or are you only using what we already have? In case you didn't add code, I recommend not using the lane detection operator as it is not currently fully implemented. Pylot's planner does not use the detected lanes, and instead relies on the HD map (you can see that lanes are not used here: https://github.com/erdos-project/pylot/blob/master/pylot/planning/world.py#L29).

Nonetheless, it would be useful if you can provide instructions on how to reproduce this error or the location values the `transform_locations` method on line 190 is invoked with. This would help us debug, and fix this before we finish integrating lane detection in the planner.

FitashUlHaq, 2021-06-14:
Thank you @ICGog for replying, actually I started using e2e.config file in the start and it gave me lanenet_model not found error. I opened this [`issue`](https://github.com/erdos-project/pylot/issues/154#issue-788372709) and it helped me get rid of that error.

How can I turn lanenet detection off from config file?

this is the config file for roundabout
--simulator_town=3
--simulator_spawn_point_index=122
--goal_location=1.51742523, -22.96622559, 0.275307

ICGog, 2021-06-14:
You can turn it off by commenting `--lane_detection` in the config file (https://github.com/erdos-project/pylot/blob/master/configs/e2e.conf#L26).

FitashUlHaq, 2021-06-14:
Thank you very much!

ICGog, 2021-06-19:
Closing this issue because I haven't managed to reproduce it despite running with the provided configuration.

closed 2021-06-19 by ICGog