erdos-project/pylot #194: CARLA version 784d9b9f is not supported

bobd988, 2021-06-13:
Hello,

I pulled the latest from master and followed the instructions to run manually but met Carla server crash issue.
PC: MSI GE66 with 64G, RTX3080
Python 3.7.10
CUDA: 11.2
CuDNN 8.1

The test steps were

 1. from terminal 1  start with command ./CarlaUE4.sh
 2. from terminal 2.  run below commands
  cd $PYLOT_HOME/scripts/
  source ./set_pythonpath.sh
  python3 pylot.py --flagfile=configs/detection.conf

After running the above command there is no error from console but Carla server disappeared after a while. I checked the pylot.log it shows:
....
...

I0613 11:31:12.184283 139961189963584 __init__.py:409] CACHEDIR=/home/demo/.cache/matplotlib
I0613 11:31:12.185990 139961189963584 font_manager.py:1468] Using fontManager instance from /home/demo/.cache/matplotlib/fontList.json
I0613 11:31:12.372965 139961189963584 component_creator.py:78] Using obstacle detector...
I0613 11:31:12.374281 139961189963584 component_creator.py:84] Adding obstacle location finder...
I0613 11:31:12.375117 139961189963584 component_creator.py:215] Using ground traffic lights from the simulator...
I0613 11:31:12.375180 139961189963584 component_creator.py:502] Not using prediction...
I0613 11:31:12.375221 139961189963584 component_creator.py:532] Using behavior planning...
I0613 11:31:12.379147 139961189963584 component_creator.py:536] Using planning...
I0613 11:31:12.383540 139961189963584 component_creator.py:573] Using the manual control/autopilot...
I0613 11:31:12.383606 139961189963584 component_creator.py:583] Synchronizing ticking using the waypoints stream
ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10

I  downloaded different versions of Carla 0.9.10, 0.9.10.1, 0.9.11 and tested separately, but all showed same error.  Has anyone see the similar issue before?

ICGog, 2021-06-13:
Hi Bob,

The ERROR you're seeing indicates that you're using a minor CARLA version. Unfortunately, CARLA doesn't set the version for minor releases (e.g., 0.9.10.1), but uses their commit number. Therefore, it's not easy for us to figure out exactly which minor CARLA version is running, and thus we default to the latest supported version 0.9.10. However, you can safely ignore this error as 0.9.10.x versions do not have API breaking changes.

Did you make sure to set `CARLA_HOME` before you ran `source ./set_pythonpath.sh`? Also, in your setup did you make sure that your `$PYTHONPATH` environment variable points to the same CARLA version as the one you're running with `./CarlaUE4.sh`? In my experience, the simulator can crash if: 1) you're using an unstable simulator version (I think I recommend using 0.9.10.4), or 2) if the CARLA version in `$PYTHONPATH` doesn't match the simulator you're actually running.

Please feel free to close this issue if my answer helped you solve the issue.

bobd988, 2021-06-14:
Hi Ionel,

I  checked CARLA_HOME and python version and make sure these were set correctly. I also tried to upgrade to Ubuntu 20.04 and downgrade python 3.6.13.  However Carla still crashed. Then I  removed the -opengl  parameter when start Carla( using the default 
version comes within pylot) with object detection and this time  Carla didnot crash and showed result correctly. However the fps is quite slow as the log says GPU was not used.  

:48:00.029573 140248497293120 component_creator.py:583] Synchronizing ticking using the waypoints stream
2021-06-14 14:48:01.305595: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcudart.so.10.0'; dlerror: libcudart.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305666: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcublas.so.10.0'; dlerror: libcublas.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305713: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcufft.so.10.0'; dlerror: libcufft.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305758: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcurand.so.10.0'; dlerror: libcurand.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305802: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcusolver.so.10.0'; dlerror: libcusolver.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305847: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcusparse.so.10.0'; dlerror: libcusparse.so.10.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305893: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcudnn.so.7'; dlerror: libcudnn.so.7: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: :/usr/local/cuda-11.1/lib64:/home/demo/Downloads/cuda/lib64
2021-06-14 14:48:01.305901: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1662] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.
Skipping registering GPU devices...
..

from log it looks for CUDA 10.0 and cudnn 7. Is there any settings to let system choose CUDA 11?  Is this the root cause for openGL crashes? Thanks.

ICGog, 2021-06-19:
It looks like Pylot is not using your GPU because your CUDA version doesn't match the version Tensorflow expects. Pylot depends on Tensorflow 1.15.4, which it needs CUDA 10.0. 

Please try to downgrade your CUDA version and let us know if that solves your issue.

bobd988, 2021-06-21:
I have downgrade to CDUA 10.0 however it crashed. It seems RTX3080 supports only CUDA 11.1+ and i guess this is the reason for crashes. I then followed this link (https://www.pugetsystems.com/labs/hpc/How-To-Install-TensorFlow-1-15-for-NVIDIA-RTX30-GPUs-without-docker-or-CUDA-install-2005/ )to have installed TF1.15 for CDUA 111. The detection demo was  started successfully, but fps is very slow. It looks  like there are still some compatibility issue for RTX3080. 
 
I guess the better way is to wait pylot to support TF2.x which will match CDUA 11.x for RTX30 hardware.  Is it a  big effort to upgrade TF2.x in Pylot ?

closed 2021-06-21 by bobd988