erdos-project/pylot #198: Error: ModuleNotFoundError: No module named 'pylot.planning.hybrid_astar.hybrid_astar_planner.HybridAStar

FitashUlHaq, 2021-06-20:
Hi 

While trying to execute hybrid_astar I am facing this error

  File "<string>", line 31, in <module>
  File "/home/erdos/workspace/pylot/pylot/planning/planning_operator.py", line 78, in __init__
    from pylot.planning.hybrid_astar.hybrid_astar_planner \
  File "/home/erdos/workspace/pylot/pylot/planning/hybrid_astar/hybrid_astar_planner.py", line 1, in <module>
    from hybrid_astar_planner.HybridAStar.hybrid_astar_wrapper \
  File "/home/erdos/workspace/pylot/dependencies/hybrid_astar_planner/HybridAStar/hybrid_astar_wrapper.py", line 12, in <module>
    from pylot.planning.hybrid_astar.hybrid_astar_planner.HybridAStar \
ModuleNotFoundError: No module named 'pylot.planning.hybrid_astar.hybrid_astar_planner.HybridAStar'; 'pylot.planning.hybrid_astar.hybrid_astar_planner' is not a package

FitashUlHaq, 2021-06-20:
Configuration used:

--random_seed=1337
###### Object detection config ######
--obstacle_detection
--nosimulator_obstacle_detection
--obstacle_detection_gpu_memory_fraction=0.2
--obstacle_detection_min_score_threshold=0.3
###### Traffic light detector config ######
--traffic_light_detection
--nosimulator_traffic_light_detection
--traffic_light_det_gpu_memory_fraction=0.2
--traffic_light_det_min_score_threshold=0.3
###### Depth #####
--perfect_depth_estimation
###### Segmentation config ######
--perfect_segmentation
###### Prediction config ######
#--perfect_obstacle_tracking
--obstacle_tracking
--tracking_num_steps=10
--prediction
#--prediction_type=linear
--prediction_num_past_steps=10
--prediction_num_future_steps=30
###### Planning config #####
--planning_type=hybrid_astar
###### Control config #####
--control=pid
#--lane_detection
--steer_gain=1.0
--stop_for_people=True
--stop_for_traffic_lights=True
--stop_for_vehicles=True
--stop_at_uncontrolled_junctions=True

FitashUlHaq, 2021-06-20:
I am using on commit ID: 388d4f77ffc12051cbc2e5966120089dba9d4138

FitashUlHaq, 2021-06-20:
Same issue with rrt_star_planning

Traceback (most recent call last):
  File "<string>", line 31, in <module>
  File "/home/erdos/workspace/pylot/pylot/planning/planning_operator.py", line 83, in __init__
    from pylot.planning.rrt_star.rrt_star_planner import RRTStarPlanner
  File "/home/erdos/workspace/pylot/pylot/planning/rrt_star/rrt_star_planner.py", line 3, in <module>
    from rrt_star_planner.RRTStar.rrt_star_wrapper import apply_rrt_star
  File "/home/erdos/workspace/pylot/dependencies/rrt_star_planner/RRTStar/rrt_star_wrapper.py", line 10, in <module>
    from pylot.planning.rrt_star.rrt_star_planning.RRTStar.py_cpp_struct \
ModuleNotFoundError: No module named 'pylot.planning.rrt_star.rrt_star_planning'

ICGog, 2021-07-23:
I assume that is happening on your local machine, and not in the Docker container. If so, then please make sure to follow the instructions and run `install.sh` on your machine. Alternatively, you can manually run the steps from https://github.com/erdos-project/pylot/blob/master/install.sh#L100 to fetch and compile the planners in the directory Pylot expects them to be.

closed 2021-07-23 by ICGog