erdos-project/pylot #203: Latest DOCKER - ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10

sigeisler, 2021-07-09:
Hi,

I use the latest docker container to collect some data (`docker run --gpus 1 -itd --name pylot -p 20022:22 erdosproject/pylot /bin/bash`). Nevertheless, I get the message:
`ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10`

Is this normal? 

Thanks,
Simon

ICGog, 2021-07-09:
In Pylot, we need to the CARLA version in order to adapt how we use the CARLA API and work around CARLA breaking changes. The log you're seeing is printed because the Docker image ships with a minor CARLA version, which is not correctly set in the CARLA release (CARLA doesn't set the version for minor releases (e.g., 0.9.10.1), but uses their commit number). Therefore, it's not easy for us to figure out exactly which minor CARLA version is running, and thus we default to the latest supported version 0.9.10. However, you can safely ignore this error as 0.9.10.x versions do not have API breaking changes.

closed 2021-07-09 by ICGog

sigeisler, 2021-07-09:
I see. Thanks!

AOOOOOA, 2025-06-20:
> In Pylot, we need to the CARLA version in order to adapt how we use the CARLA API and work around CARLA breaking changes. The log you're seeing is printed because the Docker image ships with a minor [...]

Hi, I encountered the same problem. Also, when I tried forwarding the Xauthority, no window appeared. Does this mean that Pylot is not running normally?

Thanks.