erdos-project/pylot #208: Error when running 'python3 pylot.py --flagfile=configs/detection.conf'

GarvTambi, 2021-07-21:
Traceback (most recent call last):
  File "/home/ssp/anaconda3/envs/tf/lib/python3.7/site-packages/erdos/streams.py", line 114, in internal_callback
    callback(msg, *write_streams)
  File "/home/ssp/anaconda3/envs/tf/lib/python3.7/site-packages/erdos/__init__.py", line 294, in wrapper
    return func(*args, **kwargs)
  File "/home/ssp/Downloads/pylot/pylot/perception/detection/detection_operator.py", line 126, in on_msg_camera_stream
    msg.frame.frame)
  File "/home/ssp/Downloads/pylot/pylot

avatar1912, 2021-07-22:
**this is my simulator output log:**

I0722 03:24:33.876599 140465658963776 __init__.py:409] CACHEDIR=/home/erdos/.cache/matplotlib
I0722 03:24:33.877136 140465658963776 font_manager.py:1468] Using fontManager instance from /home/erdos/.cache/matplotlib/fontList.json
I0722 03:24:34.281338 140465658963776 component_creator.py:78] Using obstacle detector...
I0722 03:24:34.281777 140465658963776 component_creator.py:84] Adding obstacle location finder...
I0722 03:24:34.282181 140465658963776 component_creator.py:215] Using ground traffic lights from the simulator...
I0722 03:24:34.282261 140465658963776 component_creator.py:499] Not using prediction...
I0722 03:24:34.282318 140465658963776 component_creator.py:529] Using behavior planning...
I0722 03:24:34.332416 140465658963776 component_creator.py:533] Using planning...
I0722 03:24:34.333755 140465658963776 component_creator.py:570] Using the manual control/autopilot...
I0722 03:24:34.333828 140465658963776 component_creator.py:580] Synchronizing ticking using the waypoints stream
2021-07-22 03:24:35.568647: E tensorflow/stream_executor/cuda/cuda_driver.cc:318] failed call to cuInit: CUDA_ERROR_UNKNOWN: unknown error
2021-07-22 03:24:35.568788: F tensorflow/stream_executor/lib/statusor.cc:34] Attempting to fetch value instead of handling error Not found: no CUDA devices found
Fatal Python error: Aborted

Current thread 0x00007fbfdffff700 (most recent call first):
  File "/home/erdos/.local/lib/python3.6/site-packages/tensorflow_core/python/client/session.py", line 699 in __init__
  File "/home/erdos/.local/lib/python3.6/site-packages/tensorflow_core/python/client/session.py", line 1585 in __init__
  File "/home/erdos/workspace/pylot/pylot/perception/detection/detection_operator.py", line 61 in __init__
  File "<string>", line 31 in <module>

Thread 0x00007fc0b5b4a740 (most recent call first):
  File "/home/erdos/.local/lib/python3.6/site-packages/erdos-0.3.1-py3.6-linux-x86_64.egg/erdos/__init__.py", line 207 in runner
  File "/usr/lib/python3.6/multiprocessing/process.py", line 93 in run
  File "/usr/lib/python3.6/multiprocessing/process.py", line 258 in _bootstrap
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 73 in _launch
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 19 in __init__
  File "/usr/lib/python3.6/multiprocessing/context.py", line 277 in _Popen
  File "/usr/lib/python3.6/multiprocessing/context.py", line 223 in _Popen
  File "/usr/lib/python3.6/multiprocessing/process.py", line 105 in start
  File "/home/erdos/.local/lib/python3.6/site-packages/erdos-0.3.1-py3.6-linux-x86_64.egg/erdos/__init__.py", line 223 in run_async
  File "pylot.py", line 219 in driver
  File "pylot.py", line 259 in main
  File "/home/erdos/.local/lib/python3.6/site-packages/absl/app.py", line 251 in _run_main
  File "/home/erdos/.local/lib/python3.6/site-packages/absl/app.py", line 303 in run
  File "pylot.py", line 272 in <module>
ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10
ERROR: CARLA version 784d9b9f is not supported; assuming this is version 0.9.10

ICGog, 2021-07-22:
According to the logs it looks like you don't have CUDA installed on your computer:

CUDA_ERROR_UNKNOWN: unknown error
2021-07-22 03:24:35.568788: F tensorflow/stream_executor/lib/statusor.cc:34] Attempting to fetch value instead of handling error Not found: no CUDA devices found

Does your machine have a GPU? Are you using the Docker image or did you install CUDA manually on your machine?

jasonleecode, 2021-07-23:
> According to the logs it looks like you don't have CUDA installed on your computer:
>
> CUDA_ERROR_UNKNOWN: unknown error
> 2021-07-22 03:24:35.568788: F tensorflow/stream_executor/lib/statusor.cc:34] [...]

yes, my machine have a GPU,  and i use docker image provide by pylot.

jasonleecode, 2021-07-23:
> > According to the logs it looks like you don't have CUDA installed on your computer:
> > CUDA_ERROR_UNKNOWN: unknown error
> > 2021-07-22 03:24:35.568788: F tensorflow/stream_executor/lib/statusor.cc:34] Attempting to fetch value instead of handling error Not found: no CUDA devices found
> > Does your machine have a GPU? Are you using the Docker image or did you install CUDA manually on your machine?
> 
> yes, my machine have a GPU, and i use docker image provide by pylot. env: ubuntu18.04 + cuda11.3 + docker

I reinstall cuda today, and resolve this problem, new env: ubuntu18.04 + cuda11.0 + cudnn8.0 + docker.
thanks for your reply.

jasonleecode, 2021-07-23:
[image]

closed 2021-07-23 by ICGog