erdos-project/pylot #212: Test the agent's performance in the docker

estherxpr, 2021-08-17:
I am wondering if there is a way to test the agent's performance inside the container instead of uploading the image to the CARLR leaderboard? For some reason, our CARLA leaderboard status is still pending so we cannot upload the image to the benchmark.

ICGog, 2021-08-20:
The Docker image already has installed the leaderboard repository. You should be able to test the agent on the training routes by executing ```cd /home/erdos/workspace/leaderboard ; ./scripts/run_experiments.sh``` in the container.

estherxpr, 2021-08-20:
Thanks! But when I tried to run it, it gives back the error: '/home/ionel/code/challenge/leaderboard//leaderboard/leaderboard_evaluator.py': [Errno 2] No such file or directory

estherxpr, 2021-08-20:
Okay! I am so sorry that I still encounter weird issue: 
If I do:
erdos@d62bb4c72b37:~$ cd /home/erdos/workspace/leaderboard
erdos@d62bb4c72b37:~/workspace/leaderboard$ ./scripts/run_evaluation.sh

I got:
python3: can't open file '/leaderboard/leaderboard_evaluator.py': [Errno 2] No such file or directory