erdos-project/pylot #217: Link to download lane detection model is not valid

Chiakiaki, 2021-09-26:
Hi,

It seems that the link to download lane detection models:
https://drive.google.com/uc?id=1yILceTgUr1MTZ7Q2wanvQ0r8RbBf5ILD
is not valid not.

Could you provide a new link to download the lane detection models? Thank you very much.

ICGog, 2021-09-29:
Hi @Chiakiaki ,

We were in the process of updating our weights to TensorFlow2, but we have finished the update in PR https://github.com/erdos-project/pylot/pull/220. We're planning on doing a new Pylot and Docker image release later today. 

Please pull the v0.3.3 Pylot image and let us know if you still run into issues.

Chiakiaki, 2022-02-11:
Hi, @ICGog 

Long time since the last post. I pulled the v0.3.3 pylot image, but got many errors like:
"GPU:0 failed, out of memory." and "pop from an empty queue"
while running:
python3 pylot.py --flagfile=configs/detection.conf
and:
python3 pylot.py --flagfile=configs/demo.conf
May I know what is the required gpu memory to run these demos? (My PC has a 8gb nvidia gpu)

I also wish to reproduce your result in recent Carla Challenge, so I also pulled your challenge branch image. But I'm not quite sure how to run it. The Carla challenge only says we can run it locally, but it didn't say how. Could you tell me something about running your challenge branch image? E.g. if I like mount local Carla home directory into the docker container, and set some env variables, is that enough?

closed 2022-02-23 by Chiakiaki

ry4nzhu, 2023-09-04:
> n3 pylot.py --fla

Hi, I am wondering if you have solved this issue. I am encountering the same GPU out-of-memory issue. Can you advise me on that?

pschafhalter, 2023-09-06:
Hi, we typically run using an NVIDIA Titan RTX with 24 GB GPU memory.

The exact memory requirements depend on the Pylot configuration. You reduce GPU memory requirements by using smaller models and decreasing sensor resolutions (e.g., by reducing `--camera_image_width` and `--camera_image_height`).

One way to test if Pylot will run on your system is to run `perfect_detection.conf`, which doesn't execute any models on the GPU. This should check if you have enough GPU memory to run CARLA with the default 1280x720 camera sensor configuration.