erdos-project/pylot #232: Failed to connect with simulator when testing agent

panpan-77, 2022-03-15:
Hello,
Thank you for your work.
I am very interested in your project.
I try to test the agent's performance. When I run ./scripts/run_evaluation.sh. It has the following error. It looks like the connection to the simulator failed. 

[image]
But I have run the simulator locally.
[image]

Did I miss anything? 
Thank you in advance!

klaus99815, 2022-03-16:
> It looks lie you're running the simulator outside of the container. Therefore, Pylot, which is running in the container, cannot establish a connection to the simulator. Please run the simulator inside the container by executing: `nvidia-docker exec -i -t your_container_name /home/erdos/workspace/pylot/scripts/run_simulator.sh`.

Thank you for your response.
But there is a problem when I run nvidia-docker exec -i -t pylot /home/erdos/workspace/pylot/scripts/run_simulator.sh in docker. 
[image]
In another terminal,It seems that the connection to the simulator has failed again.

I tried the suggestions you mentioned here：
https://github.com/erdos-project/pylot/issues/210#issuecomment-898588901
It doesn't work for me. I also tried other methods mentioned in this issue, which didn't work.

I run the run_simulator.sh in the pylot container and run_evaluation.sh in the pylot-challenge container. Is that right?

I would appreciate it if you could provide some advice!
Thanks again!

panpan-77, 2022-03-16:
> It looks lie you're running the simulator outside of the container. Therefore, Pylot, which is running in the container, cannot establish a connection to the simulator. Please run the simulator [...]

@ICGog ,Thank you for your response.
But there is a problem when I run nvidia-docker exec -i -t pylot /home/erdos/workspace/pylot/scripts/run_simulator.sh in docker. 
[image]
In another terminal,It seems that the connection to the simulator has failed again.

I tried the suggestions you mentioned here：
https://github.com/erdos-project/pylot/issues/210#issuecomment-898588901
It doesn't work for me. I also tried other methods mentioned in this issue, which didn't work.

I run the run_simulator.sh in the pylot container and run_evaluation.sh in the pylot-challenge container. Is that right?

I would appreciate it if you could provide some advice!
Thanks again!