erdos-project/pylot #246: Implement without Carla

yuanhaorannnnnn, 2022-04-08:
Hello community,
According to the introduction, pylot is suitable for Carla and real-world, so I wonder how to run pylot app without Carla simulator.
For example, I get all the inputs from the various sensors and then put them into models to get some detections results, can anyone herer tell me where I can find some code example for this?
Thanks!

pschafhalter, 2022-04-08:
Hi,

We used the [`lincoln.py`](https://github.com/erdos-project/pylot/blob/master/lincoln.py) file along with the lincoln configs to run Pylot on a Lincoln MKZ autonomous vehicle. We also provide drivers for the Linocln MKZ vehicle in [`pylot/pylot/drivers`](https://github.com/erdos-project/pylot/tree/master/pylot/drivers) (for example, [`grasshopper3_driver_operator.py`](https://github.com/erdos-project/pylot/blob/master/pylot/drivers/grasshopper3_driver_operator.py) interacts with the camera).

yuanhaorannnnnn, 2022-04-08:
> Hi,
>
> We used the [`lincoln.py`](https://github.com/erdos-project/pylot/blob/master/lincoln.py) file along with the lincoln configs to run Pylot on a Lincoln MKZ autonomous vehicle. We also provide [...]

Thanks very much for your reply!

yuanhaorannnnnn, 2022-04-12:
> Hi,
>
> We used the [`lincoln.py`](https://github.com/erdos-project/pylot/blob/master/lincoln.py) file along with the lincoln configs to run Pylot on a Lincoln MKZ autonomous vehicle. We also provide [...]

Sorry for re=opening this issue. I have one more question, in the **lincoln.py**, we have to `import rospy`, does it mean we have to install all the ROS package and python bindings? Cause I didn't find relevent packages on requirements.txt
Thanks

yuanhaorannnnnn, 2022-04-13:
> Yes, the Lincoln setup relied on a full ROS installation for the drivers to interact with the sensors.

Thanks for your explanation, and now I have two small doubt.
1. **a full ROS installation** means I have to install ros desktop-full package or just a subset for exmaple **ros-base**? Besides, does pylot program have a distro version requiremeny for ros?
2. Today I've read the paper about this amazing project, I'd appreciate if you could help me verify my understanding: If I decide not to use Carla, do I must use ROS to complete the communications between nodes? Because ERDOS can not replace the ROS, am I right?

Thanks for your time and further help.

pschafhalter, 2022-04-14:
1. The full ROS installation is only needed to interface with certain sensors that require ROS to receive data. The Lincoln MKZ came with a computer that was pre-configured with ROS for interacting with the vehicle. Our provided drivers forward data between ROS topics and ERDOS streams.
2. No, ROS is not required to run Pylot. The ROS dependency is an artifact of our specific Lincoln MKZ implementation, as it allowed us to quickly port Pylot to the vehicle. A more efficient configuration that does not depend on ROS might use I/O devices to interact with the vehicle.

yuanhaorannnnnn, 2022-05-18:
Sorry for re-opening this issue because I am stucking by a problem with lincoln implementation. 
Now I can get lincoln.py script running with a real usb-camera and a hd-map data input. I'm sure you guys are very familiar with the code, so I just put my problem straight forward. 
At the end of the lincoln pipeline, there is a operator named **DriveByWireOperator**, I just found this operator is successfullly connected but not running acutally, I've double-checked the dataflow connection and confirmed the input of **DriveByWireOperator**: control_stream has sent the control message in the operator **PIDControlOperator**.
Any insight and comment? Thanks for this project and team again!

sukritkalra, 2022-05-19:
Hi @yuanhaorannnnnn!

Just for further clarification, could you let us know where exactly the `DriveByWireOperator` fails in your setup? Does it fail to initialize, or does it not seem to be receiving messages in the loop? A few simple tests to check this would be:
1. Do a `rostopic list` to see if the topics listed [here](https://github.com/erdos-project/pylot/blob/master/pylot/drivers/drive_by_wire_operator.py#L47) by the `DriveByWireOperator` are actually initialized?
2. Check if the debug statement inserted [here](https://github.com/erdos-project/pylot/blob/master/pylot/drivers/drive_by_wire_operator.py#L81) is observed in your logs.
3.  You can also `rostopic echo /vehicle/steering_cmd` to see if the commands are actually sent and if the issue is with your CAN bus' subscriber instead.

yuanhaorannnnnn, 2022-05-19:
> Hi @yuanhaorannnnnn!
>
> Just for further clarification, could you let us know where exactly the `DriveByWireOperator` fails in your setup? Does it fail to initialize, or does it not seem to be [...]

Hi @sukritkalra 
By inserting some debug statements in the **DriveByWireOperator**, I'm sure **__** **init** **__** and **connect** function is runnnig well while **run** function is not.
For your advices, here are some feedbacks

1. rostopic list
[image]
2. I didn't find this debug info because **run** function is not running
3. rostopic echo /vehicle/steering_cmd
[image]

From my ponit, the ros setup is ok so I guess there is some issues with ERDOS connection, but I'm not sure. Thanks for your further help!

sukritkalra, 2022-05-19:
Looks like the `run` method is executing fine since the topics seem to be actually created, and it seems to be publishing data on `/vehicle/steering_cmd` too since `rostopic echo` actually shows the data.

However, these seem to be the default values which means that no `control_message` was probably received from the PID Controller. Could you please attach the relevant log file so I can look at what's happening in the execution?

yuanhaorannnnnn, 2022-05-20:
> Looks like the `run` method is executing fine since the topics seem to be actually created, and it seems to be publishing data on `/vehicle/steering_cmd` too since `rostopic echo` actually shows the [...]

Thanks for your reply @sukritkalra ,here are my debug statement code at the PID Controller, I just set value manually for convenience.
[image]
and you can find a attach of log file. Many thanks.
[lincoln.log](https://github.com/erdos-project/pylot/files/8737964/lincoln.log)