erdos-project/pylot #249: [Tensorflow]Runtime error with lincoln.py

yuanhaorannnnnn, 2022-04-22:
Hello pylot community, recently I try to implement Pylot on a NVIDIA Xavier device ported a small self-driving car, thanks for previous help and it was told that using the **lincoln.py** is the easiest way to get it start without Carla. I've set all the dependencies ready including the ROS stuff. However I got a RUNTIME error that seems related with Tensorflow when I launch the lincoln.py. The errorn shows below.

> Traceback (most recent call last):
  File "<string>", line 31, in <module>
  File "/home/nvidia/pylot/pylot/perception/detection/detection_operator.py", line 63, in __init__
    self.__run_model(np.zeros((108, 192, 3), dtype='uint8'))
  File "/home/nvidia/pylot/pylot/perception/detection/detection_operator.py", line 161, in __run_model
    infer = self._model.signatures['serving_default']
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/saved_model/signature_serialization.py", line 213, in __getitem__
    return self._signatures[key]
KeyError: 'serving_default'

**What I've done:**

- I logged the model's signatures'keys, it's empty;

- I replaced the model with others also git-cloned with origin pylot project but nothing lucky.

I digged a lot on the stackoverflow and some Tensorflow issue discussions but nothing found. So I post it here and hope any help for me to get a solution.

pschafhalter, 2022-04-26:
It looks like there might be errors in loading the model. Could you try loading a model outside of Pylot? These should be located in `pylot/dependencies/models/obstacle_detection`.
```python
model = tf.saved_model.load(model_path)
print(list(model.signatures.keys()))
```

yuanhaorannnnnn, 2022-04-27:
> It looks like there might be errors in loading the model. Could you try loading a model outside of Pylot? These should be located in `pylot/dependencies/models/obstacle_detection`.
> 
> ```python
> model = tf.saved_model.load(model_path)
> print(list(model.signatures.keys()))
> ```

After a quick test, I thinks maybe there is somthing wrong with the model. cause I loaded the model in python interpretor and  got the same result, the signatures is empty as well.

yuanhaorannnnnn, 2022-04-27:
Finally I change the model as SSD insted, for now it wokts fine.

closed 2022-04-27 by yuanhaorannnnnn