erdos-project/pylot #250: ValueError: Unknown CUDA arch (8.6) or GPU not supported

wutianze, 2022-04-24:
When trying to install pylot in my base system, I came up with the following error.
`/home/sauron/.local/lib/python3.8/site-packages/torch/include/ATen/core/TensorBody.h:322:7: note: declared here
  322 |   T * data() const {
      |       ^~~~
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    setup(
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 148, in setup
    return run_commands(dist)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
    dist.run_commands()
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/command/build.py", line 136, in run
    self.run_command(cmd_name)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/dist.py", line 1214, in run_command
    super().run_command(command)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/home/sauron/.local/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/home/sauron/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 372, in build_extensions
    build_ext.build_extensions(self)
  File "/home/sauron/.local/lib/python3.8/site-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/home/sauron/.local/lib/python3.8/site-packages/setuptools/_distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/sauron/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 288, in unix_wrap_compile
    "'-fPIC'"] + cflags + _get_cuda_arch_flags(cflags)
  File "/home/sauron/.local/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1027, in _get_cuda_arch_flags
    raise ValueError("Unknown CUDA arch ({}) or GPU not supported".format(arch))
ValueError: Unknown CUDA arch (8.6) or GPU not supported
[x] Downloading the CARLA 0.9.10.1 simulator...
`
My GPU is rtx3090ti, and the cuda version is 11.6

wutianze, 2022-04-24:
The problem is:

pschafhalter, 2022-04-26:
Does this problem also occur when running Pylot using the docker container?

If you don't need to use Pylot with AnyNet, you could also try commenting out the following lines of the installation script and re-running:
https://github.com/erdos-project/pylot/blob/765bbfa59812006961f5868bcb0a240cc6770b98/install.sh#L100-L104

wutianze, 2022-04-26:
> Does this problem also occur when running Pylot using the docker container?
>
> If you don't need to use Pylot with AnyNet, you could also try commenting out the following lines of the installation [...]

Thanks for your reply. This problem would not occur when running docker container since the container simulates a cuda 10 environment. 
And I will try commenting out AnyNet, thanks again. By the way, I wonder if it is possible to use UE4 to make some changes to the pylot scenario(the city staff) based on your containers instead of using the base system.

pschafhalter, 2022-04-26:
Pylot is bundled with the [CARLA simulator](http://carla.org/). We provide some flags for configuring the simulator (e.g. # of vehicles and pedestrians, weather, which town is used -- run `python3 pylot.py --helpfull` for details), but you may need modify CARLA directly depending on your requirements.

wutianze, 2022-04-26:
> Pylot is bundled with the [CARLA simulator](http://carla.org/). We provide some flags for configuring the simulator (e.g. # of vehicles and pedestrians, weather, which town is used -- run `python3 [...]

Yes, I will try modifying CARLA in the container you provided, no worrying about building a CARLA environment again, hope that will work.

closed 2022-04-26 by pschafhalter