erdos-project/pylot #251: Run pylot in the base system fails

wutianze, 2022-04-24:
Follow the installation guide in https://pylot.readthedocs.io/en/latest/manual_installation.html. I try to run pylot in my base system. The simulation is started successfully, howerver, when running pylot using `python3 pylot.py --flagfile=configs/detection.conf`, error   shows `AttributeError: module 'erdos' has no attribute 'Operator'`.

pschafhalter, 2022-04-26:
Could you check that the detection operator finishes initializing before the callback is invoked? One way to do this is by adding a print statement after the following lines to ensure that the model is loaded:
https://github.com/erdos-project/pylot/blob/765bbfa59812006961f5868bcb0a240cc6770b98/pylot/perception/detection/detection_operator.py#L54-L55

wutianze, 2022-04-26:
> Could you check that the detection operator finishes initializing before the callback is invoked? One way to do this is by adding a print statement after the following lines to ensure that the model [...]

Thanks for your reply. The problem happens because the installation is not complete, as a result, the model is not initialized correctly.

JackWetherell, 2022-04-26:
@wutianze I also have the issue. Can you please describe `the installation is not complete` a little more? What additional steps did you take to fix this?

wutianze, 2022-04-26:
> @wutianze I also have the issue, and I am using the pre-built docker. Can you please describe `the installation is not complete` a little more? What additional steps did you take to fix this?
> 
> @pschafhalter Is it possible the models aren't included in the docker image?

I tried building pylot in my base system and the build process is not finished correctly because 30 series GPU doesn't support CUDA 10(you can check this issue: https://github.com/erdos-project/pylot/issues/250).
However, since you are using the docker container, it should work without error
I will reopen this issue for those who meet the same problem.

JackWetherell, 2022-04-27:
@wutianze thank you.

I have determined why I was getting this issue for my particular use case in the docker container.

I was using the config: `pylot/simulation/challenge/challenge.conf`, and this contains some out-of-date paths for models (possibly the paths are wrong in some other configs too?)

An example path is: `--obstacle_detection_model_paths=/workspace/team_code/dependencies/models/obstacle_detection/faster-rcnn/frozen_inference_graph.pb`

And changing this to `/home/erdos/workspace/pylot/dependencies/models/obstacle_detection/faster-rcnn/frozen_inference_graph.pb` fixed the issue. 

@pschafhalter Is this worth me fixing and raising a PR, also making the error a little more clear that it is the filepath that is wrong, as the current error does not indicate this.

pschafhalter, 2022-04-28:
@JackWetherell `challenge.conf` is designed for the [CARLA Challenge docker container](https://github.com/erdos-project/pylot/#carla-autonomous-driving-challenge) which has a different directory layout. A PR that improves the error messages and comments `challenge.conf` to clarify this would be very welcome, but I wouldn't update the paths.

jingzhengli, 2024-05-21:
> I think I found the main problem. Erdos 0.4.0 changed its API and is incompatible with Pylot.

How to solve this error. I met the same error

pschafhalter, 2024-05-21:
Install an earlier version of erdos using pip with `pip install erdos==0.3.1`, or `git checkout` a tag corresponding to a prior release when installing from source.