erdos-project/pylot #254: Use HDMAP directly from opendrive without Carla

yuanhaorannnnnn, 2022-04-27:
Hello,
I want to figure out how to use the hdmap exported from openstreet without Carla. I've chekced the code **hd_map.py** in **map** directory, and found it deeply binded with Carla.
Now I have two questions:

1. How to read opendrive map data(.osm file) directly in Pylot. From my intuition it would be use **fopen("map.osm")**, but It must have someway else, right? Because I noticed the code **open_drive_stream = erdos.IngestStream()** in **lincoln.py**, it means it could use ERDOS to read the opendrive string data, I dont't know how to make driver sends data with IngestStream, hope someone could tell me the solution.
2. How to use the hdmap, if question one is solved, I can read the opendrive stream, for now I guess I need to parser the map from opendrive format to the plannnig and localization model's requirement. Would anyone inform me for this?

Thnaks for everyone here, I've worked on pylot for a while, I found it's a great project to explore and research the self-driving program, thanks again for any help.

pschafhalter, 2022-04-28:
1. We haven't worked with real-world HD mapping much, but you could try reading the OpenDRIVE map data and sending the string on the `IngestStream`. This works for running a vehicle in small regions (e.g. CARLA scenarios); however, I could also imagine developing an HDMap operator which takes in the pose stream and sends out updated maps when the vehicle moves to a new section of the map.
2. I believe the planner and the localization module should automatically parse maps in the OpenDRIVE format. `hd_map.py` uses CARLA's map object to parse and interact with arbitrary OpenDRIVE maps, so the CARLA integration should not pose a problem.

yuanhaorannnnnn, 2022-05-17:
> 1. We haven't worked with real-world HD mapping much, but you could try reading the OpenDRIVE map data and sending the string on the `IngestStream`. This works for running a vehicle in small regions [...]

Thanks for your reply and I've been working on this, and now I need your help. In the **lincoln.conf**, this is a waypoint csv file item. I found my pipeline was stuck may because I didn't inject the waypoint file, but I dont know how to generate a file like this, so could you send me a hard copy of waypoint csv file?

sukritkalra, 2022-05-17:
> Thanks for your reply and I've been working on this, and now I need your help. In the **lincoln.conf**, this is a waypoint csv file item. I found my pipeline was stuck may because I didn't inject [...]

Unfortunately, the CSV file of waypoints is dependent on the actual real-world route you want the vehicle to take, so sharing it would not really be helpful to running the vehicle. However, the CSV file is a simple list of `(X, Y, Z)` coordinates that you want the vehicle to follow, and you can look at how we parse such a file [here](https://github.com/erdos-project/pylot/blob/388d4f77ffc12051cbc2e5966120089dba9d4138/pylot/planning/waypoints.py#L24). It should be pretty easy to generate such a file from the HD-Map.