erdos-project/pylot #261: erdos module error during the first run

fabiom95, 2022-05-11:
`erdos` module gives an error when i follow the manual installation instructions listed here to run pylot.py:
```
export CARLA_HOME=$PYLOT_HOME/dependencies/CARLA_0.9.10.1/
cd $PYLOT_HOME/scripts/
source ./set_pythonpath.sh
cd  $PYLOT_HOME/
python3 pylot.py --flagfile=configs/detection.conf
```
The output is:

```
I tensorflow/stream_executor/platform/default/dso_loader.cc:53] Successfully opened dynamic library libcudart.so.11.0
Traceback (most recent call last):
  File "pylot.py", line 261, in main
    node_handle, control_display_stream = driver()
  File "pylot.py", line 51, in driver
    pipeline_finish_notify_stream,
  File "/home/fabio/Desktop/pylot/pylot/pylot/operator_creator.py", line 15, in add_simulator_bridge
    from pylot.simulation.carla_operator import CarlaOperator
  File "/home/fabio/Desktop/pylot/pylot/pylot/simulation/carla_operator.py", line 20, in <module>
    class CarlaOperator(erdos.Operator):
AttributeError: module 'erdos' has no attribute 'Operator'`

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "pylot.py", line 274, in <module>
    app.run(main)
  File "/home/fabio/anaconda3/envs/pylotEnv_py37/lib/python3.7/site-packages/absl/app.py", line 312, in run
    _run_main(main, args)
  File "/home/fabio/anaconda3/envs/pylotEnv_py37/lib/python3.7/site-packages/absl/app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "pylot.py", line 269, in main
    shutdown_pylot(node_handle, client, world)
UnboundLocalError: local variable 'node_handle' referenced before assignment
```

It seems I am having problems with the `erdos` module.
I installed the version 0.4.0 and I checked it is listed in the virtual environment. (I tried also other versions as the 0.3.1).

sukritkalra, 2022-05-17:
Hi @fabiom95!

ERDOS v0.4.0 has a new API that isn't compatible with the Pylot API (we are currently in the midst of porting Pylot to it and you can follow the development on the [`redesign`](https://github.com/erdos-project/pylot/tree/redesign) branch.

However, Pylot should be compatible with v0.3.1, but it looks to me that your `erdos` installation didn't complete successfully. Did you build it from Github or do a `pip install erdos==0.3.1`? 

In any case, could you please try running any of the examples from the ERDOS repository [here](https://github.com/erdos-project/erdos/tree/v0.3.1/python/examples) (maybe the [watermarks.py](https://github.com/erdos-project/erdos/blob/v0.3.1/python/examples/watermarks.py) one) and let us know what errors you receive if any?

Also, you could pull the `pylot` docker image to ensure that all your dependencies are in sync, which is the current recommended way of experimenting with pylot.

yang-yk, 2022-07-19:
Hello, I have met the same problem. Have you solved it?  I have changed my erdos version to 0.3.1 and the problem still exists. The code stopped in the `driver` function in the pylot.py file [(line 55 for adding sensors).](https://hub.fastgit.xyz/erdos-project/pylot/blob/master/pylot.py#L55) Thanks very much! @[fabiom95](https://hub.fastgit.xyz/fabiom95) @[sukritkalra](https://hub.fastgit.xyz/sukritkalra)