erdos-project/pylot #273: attributeError in CocoDetectionScoringModule

lizhuo-1994, 2022-06-23:
Hi Dear Pylot team,

I happened to meet the following errors when I was trying to evaluate a obstacle detection.

I cannot find self.from_dataset or from_dataset over other place in the repo.

Traceback (most recent call last):
  File "/home/lizhuo/anaconda3/envs/pylot/lib/python3.8/site-packages/erdos/__init__.py", line 253, in internal_watermark_callback
  File "/home/lizhuo/anaconda3/envs/pylot/lib/python3.8/site-packages/erdos/__init__.py", line 253, in internal_watermark_callback
    callback(timestamp, *write_streams)
  File "/home/lizhuo/anaconda3/envs/pylot/lib/python3.8/site-packages/erdos/__init__.py", line 294, in wrapper
    callback(timestamp, *write_streams)
  File "/home/lizhuo/anaconda3/envs/pylot/lib/python3.8/site-packages/erdos/__init__.py", line 294, in wrapper
    return func(*args, **kwargs)
    return func(*args, **kwargs)
  File "/home/lizhuo/pylot/pylot/perception/base_perception_eval_operator.py", line 119, in on_watermark
    self.__drain_accuracy_compute_buffer(game_time)
  File "/home/lizhuo/pylot/pylot/perception/base_perception_eval_operator.py", line 119, in on_watermark
    self.__drain_accuracy_compute_buffer(game_time)
  File "/home/lizhuo/pylot/pylot/perception/base_perception_eval_operator.py", line 126, in __drain_accuracy_compute_buffer
    if et <= up_to_time:
  File "/home/lizhuo/pylot/pylot/perception/base_perception_eval_operator.py", line 127, in __drain_accuracy_compute_buffer
    self.compute_accuracy(st, et, end_anchored)
TypeError: '<=' not supported between instances of 'NoneType' and 'int'
  File "/home/lizhuo/pylot/pylot/perception/base_perception_eval_operator.py", line 147, in compute_accuracy
    self.scoring_modules[anchor_type].add_datapoint(
  File "/home/lizhuo/pylot/pylot/perception/detection/detection_eval_operator.py", line 90, in add_datapoint
    [obstacle_to_dict(go)
  File "/home/lizhuo/pylot/pylot/perception/detection/detection_eval_operator.py", line 91, in <listcomp>
    for go in ground_obstacles if go.label in self.from_dataset]
AttributeError: 'CocoDetectionScoringModule' object has no attribute 'from_dataset'

JamesLuyangJiang, 2023-07-23:
Same issue here, is it fixed by them yet? If not, how did you get around it yourself? Thanks!

lizhuo-1994, 2023-07-23:
> @lizhuo-1994 Same issue here, is it fixed by the pylot team yet? If not, how did you get around it yourself? Thanks!

Not yet, I moved to another project...