erdos-project/pylot #281: IndexError:pop from an empty deque

zxkic, 2022-10-19:
Hello, I run the simulator successfully. However, when I try to run pylot.py by`python3 pylot.py --flagfile=configs/mpc.conf`, Error raises`IndexError:pop from an empty deque`. The pylot window shows black screen,like photo.
[image]
[image]
Thank you for your help

pschafhalter, 2022-10-21:
Hi, I unfortunately wasn't able to reproduce the error. The window should display a feed of the front-facing camera along with waypoints. Can you make sure that you have enough GPU memory available, e.g. by running `nvidia-smi`?

Also, it looks like you've manually installed Pylot. We recommend running Pylot using the docker container (see the README) -- could you see if the issue also occurs there?

zxkic, 2022-10-22:
Thank you for your response! My server has four 3090ti and I think It works well. 
[image]
And I am sorry to say that My server runs many errors in the docker version. The CARLA can't work in the docker.

pschafhalter, 2022-10-22:
I see, that should be more than powerful enough. The docker container has trouble with 30-series GPUs because they use a newer CUDA version.

Are the vehicle status numbers in the top left corner changing? If they are, Pylot might actually be running ok, and there's some kind of pygame/CARLA rendering issue. If the index error only occurs once, Pylot was probably able to recover and resume normal operations. We've tested with `pygame==1.9.6` and CARLA 0.9.10.1.

zxkic, 2022-10-25:
My environment is satisfied with pygame==1.9.6 and CARLA 0.9.10.1. I think the pylot may run well but it cannot provide the screen information. I will try pylot in the 20-series server. Thank you for your help!

closed 2022-10-31 by zxkic