erdos-project/pylot #284: how to get the conf file setting manual or documentation

aming439836, 2022-11-07:
As you know , we need run the pylot by the flagfiles ? but I don't know how to  set and make a new flagfile , and can you show how to set the flagfile according to myself requirement ,and can you give me some manul or documentation which explain the setting in the flagfile ? 

thank you so much ,and looking forwards for your reply.

aming439836, 2022-11-07:
for example :
[image]
how to get other flags setting in the above picture ?

pschafhalter, 2022-11-10:
Hi,

Example flag files are available in the [`./configs`](https://github.com/erdos-project/pylot/tree/master/configs).

You can modify these, or use them to start your own!

closed 2022-11-10 by pschafhalter