erdos-project/pylot #291: Does Pylot support Carla Town11 (or any Large map)?

Morphlng, 2023-04-26:
We've modified pylot a bit to support Carla 0.9.13 (and hopefully later version). It has been tested on the maps that came along the official distribution. However, when testing on Town11 (https://carla.readthedocs.io/en/0.9.13/#large-maps), Pylot does not work properly.

From the console output, Pylot successfully connected to carla and find the Ego vehicle we generated. But then, the spectator of the world attached to a wrong place, and the ego vehicle does not move at all.

The concept of Large map is different from other Town, there maybe a different behavior at all. Does you guys or any user meet the same problem? Is it possible to resolve?

[image]

pschafhalter, 2023-05-04:
We haven't tested Pylot on large maps, but it should be possible with modifications.

You will need to modify Pylot's `spawn_ego_vehicle` method by adding `v_blueprint.set_attribute('role_name', 'hero' )`. This ensures that CARLA loads map tiles near the ego vehicle and doesn't mark the ego vehicle as a dormant actor.
https://github.com/erdos-project/pylot/blob/a71ae927328388dc44acc784662bf32a99f273f0/pylot/simulation/utils.py#L192-L213

Morphlng, 2023-05-04:
Thanks for the reply, but this is not exactly the problem I'm facing. We're using Pylot to control the ego vehicle in our custom scenario, thus all the vehicles are spawned before Pylot takes over.

The problem is that Pylot doesn't seem to have a valid path planning, I've wrote a minimal reproducible script to generate an ego only scenario:

```python
def main():
    client = carla.Client("localhost", 2000)
    client.set_timeout(10.0)
    world = client.load_world("Town11")
    map = world.get_map()
    blueprint_library = world.get_blueprint_library()
    spawn_points = map.get_spawn_points()
    
    # Spawn hero vehicle
    hero_bp = blueprint_library.filter("model3")[0]
    hero_bp.set_attribute("role_name", "hero")
    hero_transform = random.choice(spawn_points)
    hero_vehicle = world.spawn_actor(hero_bp, hero_transform)
```

After this script end, start Pylot with: 
```bash
docker exec -it pylot bash -c 'source /home/erdos/.bashrc;python3 pylot.py --flagfile=configs/scenarios/person_avoidance_frenet.conf --simulator_host=172.17.0.1 --simulator_port=2000 --goal_location=-238.007019,-4065.660645,174.704025'
```

After Pylot found the ego vehicle, it starts to control. However, at each step it just steer to either left/right, and not pushing the throttle at all. `ControlMessage` line in below screenshot shows the exact behavior:

[image]

We've suspect that this can be related to the configuration file, so I'll put it here too:

```yaml
--scenario_runner
--simulator_mode=synchronous
--simulator_timeout=120
--throttle_max=1.0
--simulator_fps=20
--random_seed=1337
###### Perception config #####
--simulator_obstacle_detection
--simulator_traffic_light_detection
--perfect_obstacle_tracking
--tracking_num_steps=5
######### Prediction config #########
--prediction
--prediction_type=linear
--prediction_num_past_steps=5
--prediction_num_future_steps=40
--dynamic_obstacle_distance_threshold=70
--static_obstacle_distance_threshold=70
######### Planning config #########
--planning_type=frenet_optimal_trajectory
--target_speed=8
--max_speed=12
--d_road_w=0.3
--dt=0.3
--obstacle_clearance_fot=0.5
--maxt=8.0
--mint=2.0
--d_t_s=0.25
--ko=100
--max_curvature=1.0
--max_accel=6.0
--max_road_width_l=5.0
--max_road_width_r=1.0
--num_waypoints_ahead=100
--obstacle_radius=1
--goal_location=-238.007019,-4065.660645,174.704025
######### Control config #########
--control=pid
--stop_for_vehicles=False
--stop_for_people=False
--stop_for_traffic_lights=False
--steer_gain=1.0
###### Evaluation #####
--evaluation
```

Morphlng, 2023-05-04:
Ok, I've changed the `frenet_optimal_trajectory` to `hybrid_astar`, and Pylot successfully starts to drive towards the goal. So the problem is indeed related to a planning issue

pschafhalter, 2023-05-04:
Could you try setting `--planning_type=waypoint`? We typically use this configuration which makes some additional changes to run the Frenet Optimal Trajectory planner.

Morphlng, 2023-05-05:
Yes, `--planning_type=waypoint` can run, but it does not avoid obstacles. And, I've solved the problem.

### 1. Reason why Ego doesn't run

I've checked the planned path given in here:

https://github.com/erdos-project/pylot/blob/a71ae927328388dc44acc784662bf32a99f273f0/pylot/planning/frenet_optimal_trajectory/fot_planner.py#L77-L112

The result shows that its planning normally, except that `remove_completed` and `remove_waypoint_if_close` never remove any waypoint. Given that the calculation of whether one waypoint should be removed is based on `(self - other).magnitude()`, I found that the cause of this problem is `z` axis.

Town11 (and Town12 perhaps) is different from other maps that ground z are close to 0, instead the map will spawn according to the location of Ego vehicle. So the planned waypoint all have z=0 as defined in this code here:

https://github.com/erdos-project/pylot/blob/a71ae927328388dc44acc784662bf32a99f273f0/pylot/planning/planner.py#L19-L39

Caused the problem of Ego not moving, because the waypoint can never reach with PIDController.

This is also the problem is following code, which is used to update the location of the spectator:

https://github.com/erdos-project/pylot/blob/a71ae927328388dc44acc784662bf32a99f273f0/pylot/simulation/carla_operator.py#L414-L419

### 2. Solution

I think the easiest solution is to use a relative z to ego transfom instead of a specific value:

```python
if self._map is not None:
    # Use the HD Map to transform a 2D location into a
    # 3D location.
    p_loc = self._map.get_closest_lane_waypoint(
        Location(x=point[0], y=point[1], z=self._world.ego_transform.location.z)).location
else:
    p_loc = Location(x=point[0], y=point[1], z=self._world.ego_transform.location.z)
```

### 3. Problem with RTT* and Hybrid_Astar

The cause of the `Fatal Error` seems to be an incompatible issue of the pre-compiled .so file. I've built again and it can run without errors. However, both of these two algorithms failed to plan when an obstacle is in front of the Ego and triggered an `emergent stop`.

pschafhalter, 2023-05-05:
Thank you for sharing your findings! Glad you were able to get it to work.

closed 2023-05-05 by pschafhalter