erdos-project/pylot #292: Is there a way to speed up Pylot's running speed?

Morphlng, 2023-05-04:
I'm wondering if there is a configuration in Pylot that can speed up the simulation speed. 

In Carla, we can use use `-benchmark` and `-fps=[fps]` when starting the server, then the simulation will go to the next frame whenever the server receives a control package.

e.g. you set `-fps=20`, and you send the control package 40 times per second. You are actually having a real-time world running at twice of the speed.

As I understood, the key is to give control more frequently in order to have this kind of speed up. I've tried the `--simulator_control_frequency` mentioned in document, but it doesn't seem to work. Am I using the wrong config or Pylot does not provide such mechanism?

pschafhalter, 2023-05-04:
Pylot is able to execute at least as fast at real time.

We found that the bottleneck is CARLA's rendering. For example, some of our [experiments](https://github.com/erdos-project/erdos-experiments) from the D3 paper take several days to run. Some ways to speed up the rate of simulation include:
- Reducing the number of sensors. Out-of-the-box, Pylot spawns 2 cameras for driving and an [optional LiDAR](https://github.com/erdos-project/pylot/blob/2b3ad86d3782e2f931089da0d0f03d2c060a0046/pylot/flags.py#L225-L232). Making sure that the LiDAR isn't spawned can improve the simulation speed.
- Disabling visualization. Pylot spawns additional cameras for visualization, which further reduces the simulation speed.
- Reducing sensor quality, e.g. by setting `--camera_image_width=1280` and `camera_image_height=720` (defaults to 1920x1080). Higher quality sensors also take longer to render.

Morphlng, 2023-05-08:
Thanks for the reply and I've tested these tricks already. It does seem to help improve the performance, especially reducing the sensor quality.

But I'm looking forward to a way to simulate faster (e.g. twice the speed of real-time). I saw that in https://github.com/erdos-project/frenet_optimal_trajectory_planner, there is a comparison between multi-thread and single thread. I'm wondering if this is enabled by default? Does other module support this kind of performance increment as well?

pschafhalter, 2023-05-10:
We haven't implemented a flag for multi-threaded planning, but you can increase the number of threads available by editing the following line:
https://github.com/erdos-project/pylot/blob/a71ae927328388dc44acc784662bf32a99f273f0/pylot/planning/frenet_optimal_trajectory/fot_planner.py#L23

pschafhalter, 2023-05-10:
From our analysis, most of the time is spent in perception and planning. We haven't implemented performance optimizations for other modules; however, you might be able to speed up the perception components by using ML compilers like [XLA](https://www.tensorflow.org/xla), [TVM](https://tvm.apache.org/), or [`torch.compile`](https://pytorch.org/tutorials/intermediate/torch_compile_tutorial.html) to compile the models.

Morphlng, 2023-05-20:
Thank you very much for all the tips, but I'm getting only a slight improvement really. I guess Pylot is just too heavy and its quite hard to make it run fast while still maintain the performance. I will try to remove some module in our workflow to further investigate this issue.

closed 2023-05-20 by Morphlng